bambulab/BambuStudio #11: Numbers do not fit within settings boxes on Windows (10/11)

RobinWhitfield, 2022-07-16:
Issue present on both 01.00.10.03 and 01.00.10.05

Quality, Strength, Speed, Support, and Others all show this issue. Non numeric settings appear to be OK.

[image]

This does not appear to be an issue on macOS.

SoftFever, 2022-07-16:
I can confirm this issue happens on my Windows 11 machine as well.

CptHungarian, 2022-07-16:
I don't have this problem and I also use Windows 11 (21H2).
I think the problem might be the custom font type the slicer uses (HarmonyOS Sans).
Just out of curiosity, what happens if you manually install the fonts from here: \resources\fonts?

SoftFever, 2022-07-16:
> 

I can confirm that installing those fonts fixed this issue.
[image]

RobinWhitfield, 2022-07-16:
> I don't have this problem and I also use Windows 11 (21H2).

What was your build process? Anything out of the ordinary?

CptHungarian, 2022-07-16:
> > I don't have this problem and I also use Windows 11 (21H2).
> 
> What was your build process? Anything out of the ordinary?
> 
> I'm guessing you didn't just manually install those fonts

I used the prebuilt version (1.00.10.03) that is available on the Discord server and I didn't have to install the fonts manually.

RobinWhitfield, 2022-07-17:
This may be a build process issue.

Closing for now as the issue is not present on the pre-built pre-release installer version of 01.00.10.05

RobinWhitfield, 2022-07-17:
Reopening this as although it's better with the pre-release build from the installer, it's clear it's still not quite right.

It looks OK until you highlight something, then it becomes clear that the numbers still don't quite fit

[image]

RobinWhitfield, 2022-07-22:
Just noting that this issue is still present in the release 01.01.00.13

RobinWhitfield, 2022-07-23:
Can this non-default font just be changed?

Some users will not have the ability to install custom fonts or applications, so will always have the above font issues when running the "portable" non-install version of Bambu Studio.

The font does not seem sufficiently different to many other fonts available as default on Windows (and other operating systems), to be worth installing it.

RobinWhitfield, 2022-08-26:
This issue was fixed in 01.01.01.01

closed 2022-08-26 by RobinWhitfield