bambulab/BambuStudio #12: Build for Linux

probonopd, 2022-07-16:
Since BambuStudio is based on PrusaSlicer, could it be built for Linux (and FreeBSD) like PrusaSlicer can?

https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

tapir, 2022-07-16:
This is kinda deal breaker. There are lots of linux users in the maker space. Could we at least get a roadmap or official plan on linux support?

probonopd, 2022-07-17:
Getting `src/nlohmann/detail/conversions/from_json.hpp:23:14: fatal error: filesystem: No such file or directory`.

What am I doing wrong?

```
[ 14%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/ArcFitter.cpp.o
[ 14%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/BoundingBox.cpp.o
[ 15%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/BridgeDetector.cpp.o
[ 15%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/FaceDetector.cpp.o
[ 15%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Brim.cpp.o
In file included from /home/runner/work/BambuStudio/BambuStudio/src/nlohmann/adl_serializer.hpp:6:0,
                 from /home/runner/work/BambuStudio/BambuStudio/src/nlohmann/json.hpp:51,
                 from /home/runner/work/BambuStudio/BambuStudio/src/libslic3r/Format/../GCode/ThumbnailData.hpp:6,
                 from /home/runner/work/BambuStudio/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:4,
                 from /home/runner/work/BambuStudio/BambuStudio/src/libslic3r/Model.hpp:18,
                 from /home/runner/work/BambuStudio/BambuStudio/src/libslic3r/FaceDetector.cpp:4:
/home/runner/work/BambuStudio/BambuStudio/src/nlohmann/detail/conversions/from_json.hpp:23:14: fatal error: filesystem: No such file or directory
     #include <filesystem>
              ^~~~~~~~~~~~
```

Full build log:
https://github.com/probonopd/BambuStudio/runs/7379382060?check_suite_focus=true#step:8:43411

Workflow (based on SuperSlicer):
https://github.com/probonopd/BambuStudio/blob/master/.github/workflows/ccpp_ubuntu.yml

Waffull, 2022-07-18:
Having no Linux build is pretty bad.  Worse yet, no road map for one.  A piece of software based on code that runs on a Linux, shouldn't be that hard to engineer.

Waffull, 2022-08-09:
Any update?  This is a pretty huge bug, leaving Linux users out.  Sitting here with a very expensive paperweight at this point!

deftdawg, 2022-08-09:
I had a go at building, but got stuck at wxWidgets trying to link against `-dep-name-NOTFOUND`...  Might be possible to get past that problem by bumping from v3.1.5 to v3.1.7 or v3.2.0... 

Since building bambu-studio seems to require building virtually all the dependencies from source in the `deps` folder and that takes a long while on my laptop, I'm going to see how functional the Windows build runs under wine.  It installs and starts at least.

[image]

Under wine, you'll also get the Mac OS known issue "setup wizard/splash screen blank on first run" from the previous version's release notes, close and reopen bambu-studio fixes the splash.

deftdawg, 2022-08-12:
Under Wine doesn't work...  Can't get past the issues I saw before.

I spent a LOT of time the past two days and finally got bambuStudio to compile and link, unfortunately it core dumps immediately after the first dialog box. :disappointed: 

[image]

```
~/source/BambuStudio/build/package/bin$ ./bambu-studio --debug 5
[2022-08-12 00:33:26.041983] [0x00007f099043c940] [trace]   Initializing StaticPrintConfigs
[2022-08-12 00:33:26.049592] [0x00007f099043c940] [info]    finished setup params, argc=3

[2022-08-12 00:33:26.049639] [0x00007f099043c940] [info]    no action, start gui directly

[2022-08-12 00:33:26.049650] [0x00007f099043c940] [info]    run, normal mode, input_files size = 0
[2022-08-12 00:33:26.049653] [0x00007f099043c940] [info]    begin to launch BambuStudio GUI soon
Segmentation fault (core dumped)
```

deftdawg, 2022-08-13:
The porting work is now basically done with my PR.

We need someone with access to the bambu labs private repo to build and integrate the `bambu_networking` plugin which is missing from this repo. Maybe @YifanWuBambu, @lanewei120 or @StoneLiBambulab can assist in making that happen?

Without it we'll be stuck unable to communicate with the printer...
```
[2022-08-13 11:45:24.198648] [0x00007ff1ab318940] [trace]   Initializing StaticPrintConfigs
loading network module at /home/deftdawg/.BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/deftdawg/.BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)
Segmentation fault (core dumped)
```

lhwei120, 2022-08-14:
> The build work is now basically done with my PR.
> 
> We need someone with access to the bambu labs private repo to build and integrate the `bambu_networking` plugin which is missing from this repo. Maybe @YifanWuBambu, @lanewei120 or @StoneLiBambulab can assist in making that happen?
> 
> Without it we'll be stuck unable to communicate with the printer...
> 
> ```
> [2022-08-13 11:45:24.198648] [0x00007ff1ab318940] [trace]   Initializing StaticPrintConfigs
> loading network module at /home/deftdawg/.BambuStudio/plugins/libbambu_networking.so
> error, dlerror is /home/deftdawg/.BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
> after dlopen, network_module is (nil)
> Segmentation fault (core dumped)
> ```

great
we will build the linux plugins soon

anyway, without the plugins, the software should work except the networking operations

tapir, 2022-08-14:
@lhwei120 above error message suggests otherwise. It might be a good idea to have libbambu_networking.so as optional to be able to still use the slicer

deftdawg, 2022-08-14:
> @lhwei120 above error message suggests otherwise. It might be a good idea to have libbambu_networking.so as optional to be able to still use the slicer

No I think he's right...  I think it's seg faulting after that, like when it's trying to paint the UI or something...  Gdb said in combobox::setfont and somewhere else when I commented out that function...  

I didn't do a debug-symbol build and I'd need to figure out how to get VS Code to step a C++ program.   

Unfortunately,  I have some happy IRL stuff to deal with for a couple of days.   I'll try to push up a podman `Containerfile` (basically same as a `Dockerfile`),  so anyone who wants to have a go building would be able to give it a try.

uncjgh, 2022-08-19:
Not having a Linux build, preferably as an applimage to be distribution agnostic, is in my opinion criminal as it ignores the fact that there are hundreds if not thousands of Linux users in the maker space.  And without access to a usable Linux build of the Studio software makes the X1 little more than a very expensive paperweight!

Bambu Lab should at least provide access to any missing deps so someone in the community could successfully create a Linux build of the Bambu Studio software.

tapir, 2022-08-20:
@deftdawg any chance we get a docker image to build it?

deftdawg, 2022-08-20:
There is now a podman [Containerfile](https://github.com/deftdawg/BambuStudio/blob/linux/Containerfile) (might also work as a Dockerfile if you rename it) in my [branch](https://github.com/deftdawg/BambuStudio/tree/linux), it can build the binary but packaging to AppImage fails due to wanting a kernel module that I don't think can be loaded inside a podman container (libfuse -- error see bottom).  @probonopd I saw you're on some AppImage threads about the libfuse thing, if you have a solution for inside podman, please let me know.

```sh
sudo podman build . -t bambu-studio # This will build, but app image creation fails on libfuse
```

If we can resolve the AppImage fuse problem and complete the build, we should be able to get the appImage off the podman container with something like this:
```sh
sudo podman run -it --rm localhost/bambu-studio /bin/bash \
  -c 'tar -c $(find build | grep ubu64.AppImage | head -1)' | tar -x
```

Alternatively, if you run the commands listed inside the Containerfile locally on an Ubuntu or Debian based-OS you should get an appImage file.

The AppImage file produced will crash (at least it does on my system) when trying to render UI stuff, hopefully someone (@tapir? @gihad ?) can figure out how to get things to build with debug symbols, then run with GDB to see where the crashes are occurring in the rendering pipeline and troubleshoot it.

Unfortunately, I don't have any more time to work on this a bit because of IRL stuff.

----
libfuse errror (fails same spot if libfuse is installed through apt with `modprobe fuse` error)
```pre
Dappimagetool-x86_64.AppImage&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.  HTTP request sent, awaiting response... 200 OK
Length: 2172096 (2.1M) [application/octet-stream]       Saving to: '../appimagetool.AppImage'                                                                           ../appimageto 100%   2.07M  8.76MB/s    in 0.2s
2022-08-20 13:08:59 (8.76 MB/s) - '../appimagetool.AppImage' saved [2172096/2172096]

 dlopen(): error loading libfuse.so.2

AppImages require FUSE to run.
You might still be able to extract the contents of this AppImage if you run it with the --appimage-extract option.       
See https://github.com/AppImage/AppImageKit/wiki/FUSE   for more information                                    

mv: cannot stat 'BambuStudio-x86_64.AppImage': No such file or directory
chmod: cannot access 'BambuStudio_ubu64.AppImage': No such file or directory                                    
/BambuStudio/build                                      
mv: cannot stat 'package/BambuStudio_ubu64.AppImage': No such file or directory
```

probonopd, 2022-08-21:
Use `../appimagetool.AppImage --appimage-extract-and-run` when you want to run it inside Podman.

deftdawg, 2022-08-21:
Thanks @probonopd,  I added some tweaks to the build_image.sh script to make it run with that arg under podman,  worked.

Here's how to build it with Podman:
```sh
# clone my branch and go into the dir
rm -rf build; sudo podman build .. -t bambu-studio  && \
sudo podman run --rm localhost/bambu-studio /bin/bash -c \
'tar -c $(find build | grep ubu64.AppImage | head -1)' | tar -xv
```
It will put the AppImage into `build` subdirectory on the host's source directory as the last step.

As for GitHub actions,  I'm open to using it,  but not sure if will work on forks.

probonopd, 2022-08-21:
> As for GitHub actions, I'm open to using it, but not sure if will work on forks.

Yes, it does work on forks.

This might get you started:
https://github.com/probonopd/BambuStudio/blob/master/.github/workflows/ccpp_ubuntu.yml

deftdawg, 2022-08-22:
Looks like I also need to figure out how to make AppImageKit bundle the following libraries to make it runable on a SteamDeck (Arch fork):

```
(deck@steamdeck bin)$ ldd bambu-studio  | grep not
        libwebkit2gtk-4.0.so.37 => not found
        libjavascriptcoregtk-4.0.so.18 => not found
```

probonopd, 2022-08-25:
All the missing libraries should be bundled from the system on which the BambuStudio AppImage is made. In fact, _all_ libraries except those blacklisted in https://github.com/AppImage/pkg2appimage/blob/master/excludelist should be bundled.

deftdawg, 2022-08-25:
Eh?  So I'm supposed to bundle **everything** in `/usr/lib` except stuff on that list?   

I feel like there should be a flag to have AppImageKit do that automatically without me needing to sort through all the system libs and cross reference them manually in a random 3rd app (we're not using `pkg2appimage` here)...  It seems like pain that everyone who builds an appimage could avoid with a little additional code upstream.

deftdawg, 2022-08-27:
[image]

Here's a basic setup to at least be able to use print models using multi-colours via the SD card from Linux...  It sucks but it should at least let you get stuff printed.

It leverages Steam to avoid needing to deal with wine directly and because it's what's available on a Steam Deck (where the root FS is read-only):

- Install Steam Client 
- Install ProtonUp-Qt

- Open Steam
	- Add Non-steam game
		- Change filter to all files
		- Navigate to Downloads
		- Select `bambu_studio_win_latest.exe`
	- find `bambu_studio_win_latest.exe` in steam library and alternate click `Properties` 
	- Under `Compatibility` tab select `Force the use of a specific Steam Play compatibility tool`
	- Choose `GE-Proton7-28`  (you may need to install this via `ProtonUp-Qt`)
	- close the `Properties` settings dialog 
	- Launch `bambu_studio_win_latest.exe`
		- this will run the installer
		- complete the install
	- find `bambu_studio_win_latest.exe` in steam library and alternate click `Properties` 
	- Under the `Shortcut` tab
		- Rename to `Bambu Studio`
		- Change `Target` to `"C:\Program Files\Bambu Studio\bambu-studio.exe"`
	- Change start location to `"C:\Program Files\Bambu Studio"`
	- close the `Properties` settings dialog 
	- find `Bambu Studio` in steam library and launch; wait for any updates to download and apply.   If there are no updates Bambu Studio should start within 30 seconds, if it does not relaunch it (sometimes it takes more then one attempt before it starts up)

Under Proton/Wine the hosts file system is mapped as a network drive under `Z:\`

You can load stl and 3mf files from `Z:\`
Export them to `Z:` and then copy them to SD cards

probonopd, 2022-08-29:
> Eh? So I'm supposed to bundle everything in /usr/lib except stuff on that list?

Of course only the ones which the application actually uses, and the ones they depend on, and so on (recursively).

> I feel like there should be a flag to have AppImageKit do that automatically without me needing to sort through all the system libs and cross reference them manually 

There are various tools like the ones on
https://github.com/AppImage/awesome-appimage/blob/main/README.md#appimage-developer-tools
that can bundle the required dependencies into AppDirs and/or AppImages.
But I guess `BuildLinuxImage.sh` in SuperSlicer is doing things "by hand" rather than using one of those, presumably to have a finer control over what exactly gets deployed. So if BambuStudio introduced new dependencies in addition to those in SuperSlicer, then their dependencies need to be deployed as well into the AppDir.
Hope this makes sense; let me know if I should elaborate.

fsnuxer, 2022-09-22:
Any news from Bambulab for an approximate release date for linux ?
( with network plugin )

lanewei120, 2022-09-23:
Hello

we have started the work of linux version
currently network plugin is not the key issue
lots of other issues needs to be solved(GUI/liveview...)

we will submit the patches as soon as we finished.
I am afraid currently we can not give a clear date here

lanewei120, 2022-09-29:
we have submited the codes which support linux basic functions.
[image]

there are still some known issues:
1. a random crash when software launch(we can ignore it and retry until success)
2. the liveview is not working

to build the linux version, I just use following commands:
`#download codes into /home/user.name/codes/bamboo_slicer, build deps into /home/user.name/codes/dep_linux
## build deps
cd /home/user.name/codes/bamboo_slicer/deps
mkdir build;cd build
cmake ../ -DDESTDIR="/home/user.name/codes/dep_linux" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1
make -j6

cd /home/user.name/codes/bamboo_slicer
mkdir build
mkdir install_dir
cd build
cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/user.name/codes/dep_linux/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release
cmake --build . --target install --config Release -j4`

if using the scripts submitted by deftdawg before
please make sure to set '-DBBL_RELEASE_TO_PUBLIC=1'

-------------------------------------------------------------------------------------------------------------------------
Here are the software lists installed before building for reference
#install cmake
sudo apt-get install cmake

#install git
sudo apt-get install git
git config --global user.name
git config --global user.email

#install g++
sudo apt-get install g++

#install opengl
sudo apt-get install build-essential libgl1-mesa-dev

#install m4
sudo apt-get install m4 

#install wayland
#check 'XDG_SESSION_TYPE' whether is wayland
sudo apt install libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules

#install pkgconf
sudo apt-get install pkgconf 

#install glu
sudo apt-get install libglu1-mesa-dev

##install cairo
sudo apt-get install libcairo2-dev

##install gtk-3-dev
sudo apt-get install libgtk-3-dev

##install libsoup
sudo apt-get install libsoup2.4-dev

##install webkit
sudo apt-get install libwebkit2gtk-4.0-dev

##install gstreamer
sudo apt-get install libgstreamer1.0-dev
sudo apt-get install libgstreamer-plugins-good1.0-dev
sudo apt-get install libgstreamer-plugins-base1.0-dev

#osmesa
sudo apt-get install libosmesa6-dev

BarrMo, 2022-10-02:
_cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/user.name/codes/dep_linux/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release
cmake --build . --target install --config Release -j4`_

After the cmake command I'm getting openvdb errors:

CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND

Seems like something is missing?

jeanblanchard, 2022-10-02:
So, I got bambu-studio to build successfully, but I get a segmentation fault each time I try to run it.

```sh
$ ./bambu-studio                                                                                                                                                 ✔ 
[2022-10-02 11:00:56.059508] [0x00007f1f9e7caec0] [trace]   Initializing StaticPrintConfigs
(process:12522): GLib-GObject-WARNING **: 11:00:56.081: invalid (NULL) pointer instance
(process:12522): GLib-GObject-CRITICAL **: 11:00:56.081: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
(process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
(process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
zsh: segmentation fault (core dumped)  ./bambu-studio
```

This is on Manjaro Linux.

I'm not too familiar with C++ build tools and debugging, so I can't pinpoint where exactly the crash occurs, but I did some digging around, and this looks very similar to a now-fixed bug in PrusaSlicer: https://github.com/prusa3d/PrusaSlicer/issues/8126. I tried a similar change to what they did, which is, replacing `wxEXEC_SYNC` by `wxEXEC_SYNC | wxEXEC_NOEVENTS` in `wxExecute` calls, but this did not change anything. Also the PS bug refers to wxWidgets 3.0, whereas BS seems to refer to wxwidgets 3.1.

PS: @BarrMo Looks like you need to install openvdb. Probably the `libopenvdb-dev` package if you're using Ubuntu.

PPS: @lanewei120's instructions mentioned the `extra-cmake-modules` dependency in the Wayland section, but you actually need it even to build for X11.

lhwei120, 2022-10-02:
> So, I got bambu-studio to build successfully, but I get a segmentation fault each time I try to run it.
> 
> ```shell
> $ ./bambu-studio                                                                                                                                                 ✔ 
> [2022-10-02 11:00:56.059508] [0x00007f1f9e7caec0] [trace]   Initializing StaticPrintConfigs
> (process:12522): GLib-GObject-WARNING **: 11:00:56.081: invalid (NULL) pointer instance
> (process:12522): GLib-GObject-CRITICAL **: 11:00:56.081: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
> (process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> (process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> (process:12522): Gtk-CRITICAL **: 11:00:56.082: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> zsh: segmentation fault (core dumped)  ./bambu-studio
> ```
> 
> This is on Manjaro Linux.
> 
> I'm not too familiar with C++ build tools and debugging, so I can't pinpoint where exactly the crash occurs, but I did some digging around, and this looks very similar to a now-fixed bug in PrusaSlicer: [prusa3d/PrusaSlicer#8126](https://github.com/prusa3d/PrusaSlicer/issues/8126). I tried a similar change to what they did, which is, replacing `wxEXEC_SYNC` by `wxEXEC_SYNC | wxEXEC_NOEVENTS` in `wxExecute` calls, but this did not change anything. Also the PS bug refers to wxWidgets 3.0, whereas BS seems to refer to wxwidgets 3.1.
> 
> PS: @BarrMo Looks like you need to install openvdb. Probably the `libopenvdb-dev` package if you're using Ubuntu.
> 
> PPS: @lanewei120's instructions mentioned the `extra-cmake-modules` dependency in the Wayland section, but you actually need it even to build for X11.

is it possible for you to use gdb to catch this stack?
with the newest codes, there is a random crash when start-up
however, it is hard to reproduce at our side now
it may be a different issue

lhwei120, 2022-10-02:
> _cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/user.name/codes/dep_linux/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" [...]

openvdb-print is not a must
it will not block the building

could you give the full log of your building?

lanewei120, 2022-10-02:
> _cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/user.name/codes/dep_linux/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" [...]

openvdb-print is not a must
it will not block the building

could you give the full log of your building?

lanewei120, 2022-10-02:
> So, I got bambu-studio to build successfully, but I get a segmentation fault each time I try to run it.
>
> ```shell
> $ [...]

is it possible for you to use gdb to catch this stack?
with the newest codes, there is a random crash when start-up
however, it is hard to reproduce at our side now
it may be a different issue

BarrMo, 2022-10-02:
> openvdb-print is not a must it will not block the building
>
> could you give the full log of your building?

[](url)
[CMakeError.log](https://github.com/bambulab/BambuStudio/files/9692983/CMakeError.log)
[CMakeOutput.log](https://github.com/bambulab/BambuStudio/files/9692984/CMakeOutput.log)

Linux Mint 20.3 Cinnamon

deftdawg, 2022-10-02:
@lhwei120 I'm a bit confused on your directions regarding `dep_linux`?  Is that a patched version of GTK3 or something?  Is it a module we can checkout from a git repo somewhere? 

@jeanblanchard @BarrMo 

You gents should submit `gdb` backtrace's (`bt`) or let us know if it looks the same as the following:
[image]

BarrMo, 2022-10-03:
> is it possible for you to use gdb to catch this stack? with the newest codes, there is a random crash when start-up however, it is hard to reproduce at our side now it may be a different issue

Weird. I sucessfully installled libopenvdb-dev. Then I ran the cmake command again and still ended up with the same error.
Thanks for replying anyway!

BarrMo, 2022-10-03:
> @lhwei120 I'm a bit confused on your directions regarding `dep_linux`? Is that a patched version of GTK3 or something? Is it a module we can checkout from a git repo somewhere?
>
> @jeanblanchard [...]

Apologies - I'm very limited with the terminal. How do I do a backtrace?

deftdawg, 2022-10-03:
> Apologies - I'm very limited with the terminal. How do I do a backtrace?

Install `gdb`, run `gdb bambu-studio`, at the `(gdb)` prompt type `run` enter, then when it crashes at the next `(gdb)` prompt type `bt`...  refer to the image I attached where I do the steps for the appimage (it's same with an extra step to mount the appimage in /tmp).

gavwhela, 2022-10-03:
Hi all,

I finally managed to get this built on Arch Linux. I had the same issue with a segmentation fault at start every time. GDB showed same stack track as was provided by @deftdawg. I found a post saying this was due to using wx functionality without initialization. Commenting out [this line](https://github.com/bambulab/BambuStudio/blob/master/src/BambuStudio.cpp#L173) which seemed to be the only wx related line prior to `Slic3r::GUI::GUI_Run` allows the application to get farther for me. Then it would crashed with an exception due to trying to create a subdirectory of a non-existent configuration directory, so I had to create `$HOME/.config/BambuStudio`. Now I have the main UI and setup wizard, but haven't tested further yet. Wanted to post in case it helps others.

To get it built on Arch was somewhat painful, I had to add a large number of additional `#include` statements, not sure what was different that made that an issue.

deftdawg, 2022-10-03:
@gavwhela I recommend ppl use Podman with the `Containerfile` to build inside a Ubuntu container, and then export it as an Appimage... The results seem to be the same built and run on Ubuntu or built on Ubuntu and run on SteamOS (Arch)...  Think we're still missing something, maybe a patch to GTK or the master?

fsnuxer, 2022-10-03:
Tried to build and run on ubuntu 22.04. Crash on startup. Got (the same)  backtrace :
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2022-10-03 16:33:51.261603] [0x00007fffed7c8940] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffec566640 (LWP 2548310)]
[Thread 0x7fffec566640 (LWP 2548310) exited]

(process:2548299): GLib-GObject-WARNING **: 16:33:51.341: invalid (NULL) pointer instance

(process:2548299): GLib-GObject-CRITICAL **: 16:33:51.341: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:2548299): Gtk-CRITICAL **: 16:33:51.342: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:2548299): Gtk-CRITICAL **: 16:33:51.342: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:2548299): Gtk-CRITICAL **: 16:33:51.342: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x00007ffff75d1cf4 in ?? () from /lib/x86_64-linux-gnu/libgtk-3.so.0
(gdb) bt
#0  0x00007ffff75d1cf4 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#1  0x00007ffff75e9664 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#2  0x00007ffff75db754 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#3  0x00007ffff6f70ec3 in g_type_create_instance () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#4  0x00007ffff6f57f4d in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#5  0x00007ffff6f591ad in g_object_new_with_properties () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007ffff6f59cb1 in g_object_new () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#7  0x00007ffff77e22d8 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#8  0x00007ffff6f70ec3 in g_type_create_instance () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#9  0x00007ffff6f57f4d in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007ffff6f591ad in g_object_new_with_properties () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff6f59cb1 in g_object_new () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00005555586c68b4 in ContainerWidget() ()
#13 0x00005555586c8fe5 in wxSystemSettingsNative::GetFont(wxSystemFont) ()
#14 0x0000555556333d53 in Label::sysFont(int, bool) ()
#15 0x0000555556333e40 in Label::initSysFont() ()
#16 0x0000555555d52bca in Slic3r::CLI::run(int, char**) ()
#17 0x0000555555cbd2d9 in main ()

```

ajacocks, 2022-10-03:
> Hi all,
>
> I finally managed to get this built on Arch Linux. I had the same issue with a segmentation fault at start every time. GDB showed same stack track as was provided by @deftdawg. I found a [...]

I managed it get it running on Arch/x64 with just this line commented out. So, definitely, this is the issue that is causing the repeated segfaults on startup.

I might take a stab at creating an AUR package for it.

Thanks for finding that!

deftdawg, 2022-10-03:
Good job gents!

https://github.com/bambulab/BambuStudio/blob/63a33c8d1a55966c36893c4f3654647083e9967b/src/BambuStudio.cpp#L173

Commenting out that line, did the trick for me as well, here is the runnable AppImage (link will expire in 14 days):
https://www.mediafire.com/file/isgckvc5qbxz22d/BambuStudio_ubu64-2022-10-03.AppImage/file

To run it, you need to do the following:
```sh
chmod +x BambuStudio_ubu64*.AppImage # make it executable after download
mkdir -p ~/.config/BambuStudio/ # Manually create the configuration directory to avoid startup crashing
LC_ALL="en_US.UTF-8" ./BambuStudio_ubu64*.AppImage # provide a default locale for distros that don't have a locale config (i.e. SteamOS/Arch sometimes needs this, Ubuntu usually doesn't)
```

Of note, the network plugin download fails, but it should be possible to export sliced files to SD and use them that way. 

I wouldn't expect us to get much further until after the Bambu Labs team get back from the week off for National day over in China, when they can push a network plugin build for Linux. (Enjoy the vacay :palm_tree: )

jeanblanchard, 2022-10-03:
I can confirm I got it to run fine too, after commenting the `initSysFont` line. (Manjaro).

Running the `bambu-studio` binary directly, the network plugin downloaded and installed fine. I could send a file to the printer, and only the webcam view did not work, as expected. Maybe the AppImage is restricting the plugin download, installation, or execution ?

Let me just say, it's much more enjoyable to use with working 3D acceleration, compared to my windows VM at 1fps :smile: .

Also, "dark mode" GTK themes do not seem to be handled great: most of the UI is light-mode only, with a few widgets being dark, like the Control inputs in the Device tab, or the Objects tree. So you might want to run with `GTK_THEME=:light` to get a consistent UI.

deftdawg, 2022-10-03:
> Running the `bambu-studio` binary directly, the network plugin downloaded and installed fine. I could send a file to the printer, and only the webcam view did not work, as expected. Maybe the AppImage is restricting the plugin download, installation, or execution ?

That's a great point, the AppImage is going to be a read-only filesystem (like a loopback mount), if the plugin is being downloaded to a path within that space (vs somewhere writable like `~/. config/BambuStudio`), then it's going to fail. 

I'll have to look into that.

probonopd, 2022-10-03:
Are these dependencies really needed? If so, they should probably be deployed into the AppImage because they cannot be expected to be "there" on all Linux distributions by default.

```
        libgstreamer-1.0.so.0
        libgstvideo-1.0.so.0
        libwebkit2gtk-4.0.so.37
        libjavascriptcoregtk-4.0.so.18
```

gavwhela, 2022-10-03:
Thanks for mentioning that the network plugin works for you @jeanblanchard! Mine wasn't working, but I took another look and in preferences I changed the "Host Setting" to "Product Host" and it worked. Thanks!

deftdawg, 2022-10-03:
@gavwhela changing this setting worked for me as well! :beers: 

I noticed if I change the `language` from Chinese to English, BBS will crash with the follow backtrace.  I expect that's because of how my SteamOS has it's locales...  Don't touch that setting and it's fine... Everything already in English anyway.

```sh
echo -e "run\nbt\nquit" | LC_ALL="en_US.UTF-8" gdb ./BambuStudio_ubu64*.AppImage 
GNU gdb (GDB) 11.2
...
(gdb) run
Starting program: /home/steam/Desktop/BambuStudio_ubu64-2022-10-03.AppImage 
...
Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::find (this=0x8, __c=95 L'_', __pos=0) at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:1247
1247    /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc: No such file or directory.
(gdb) bt
#0  std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::find(wchar_t, unsigned long) const (this=0x8, __c=95 L'_', __pos=0)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:1247
#1  0x0000555558814ad7 in wxString::Find(wxUniChar, bool) const ()
#2  0x0000555558814d00 in wxString::AfterFirst(wxUniChar) const ()
#3  0x000055555645348b in Slic3r::GUI::linux_get_existing_locale_language(wxLanguageInfo const*, wxLanguageInfo const*) ()
#4  0x0000555556454522 in Slic3r::GUI::GUI_App::load_language(wxString, bool) ()
#5  0x000055555645d3b6 in Slic3r::GUI::GUI_App::on_init_inner() ()
#6  0x000055555645f54a in Slic3r::GUI::GUI_App::OnInit() ()
#7  0x00005555587e5637 in wxEntry(int&, wchar_t**) ()
#8  0x000055555643640b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#9  0x0000555555d276cb in Slic3r::CLI::run(int, char**) ()
#10 0x0000555555c7dc89 in main ()
(gdb) exit
Undefined command: "exit".  Try "help".
(gdb) quit
```

fsnuxer, 2022-10-04:
changing line 13 in src/slic3r/GUI/Widgets/Label.cpp to a font name I have in my system get it somewhat running :
download plugin, login, see correct information on the filament loaded.
video feed not working (Media playback error: not URI handler implemented for "bambu")

UndeadKernel, 2022-10-07:
I was able to compile and run everything in Arch GNU/Linux. However, I cannot seem to install the network plugin. Whenever I try, I get a log saying:
```
[download_plugin]: check the plugin from https://api-qa.bambu-lab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.02.00.00
[download_plugin 1]: no availaible plugin found for this app version: 01.02.00.09
```

How did you guys manage to download and install the network plugin?

deftdawg, 2022-10-07:
@UndeadKernel see the previous ~5 comments about needing to change the host setting to "production host"

UndeadKernel, 2022-10-07:
@deftdawg, I see the problem: I have a Dark Theme and therefore I cannot see the settings correctly. Which one of these options is the correct one?
[image]

deftdawg, 2022-10-07:
> @deftdawg, I see the problem: I have a Dark Theme and therefore I cannot see the settings correctly. Which one of these options is the correct one?
> [image]
> 

Very bottom one

UndeadKernel, 2022-10-07:
For anyone having a dark theme, Bambu Studio can be run with a GTK light theme with the command:
`GTK_THEME=Adwaita:light ./bambu-studio`

lanewei120, 2022-10-08:
the above crash most likely is caused by the fonts missing

I have installed these two fonts under Bambu Studio/resources/fonts manually
[image]
[image]

please have a try

lanewei120, 2022-10-08:
> @lhwei120 I'm a bit confused on your directions regarding `dep_linux`? Is that a patched version of GTK3 or something? Is it a module we can checkout from a git repo somewhere?

it is just built from the deps directory of Bambu Studio
most of the sources are downloaded from internet when building
and it is the same behavior between linux&&macos&&windows

For linux, most of these deps can be installed directly

lanewei120, 2022-10-08:
> @deftdawg, I see the problem: I have a Dark Theme and therefore I cannot see the settings correctly. Which one of these options is the correct one? [image]

it seems this building comes without the flag '-DBBL_RELEASE_TO_PUBLIC=1'
there are some debug features and some unstable features displayed there,
and also it can not find the network_plugins when downloading

please add the flag '-DBBL_RELEASE_TO_PUBLIC=1' when building
and the preference dialog is like this
[image]

deftdawg, 2022-10-08:
Thanks @ lanewei120

I just did a PR that sets `-DBBL_RELEASE_TO_PUBLIC=1` for AppImage builds are created without DEBUG_BUILD_ARGS...  It also applied a couple of segfault workaround in AppRun and some other improvements that should speed up rebuilds via Podman.

deftdawg, 2022-10-08:
Lastest AppImage built from my branch is here:

https://www.mediafire.com/file/4dbb1k21wsinhcg/BambuStudio_ubu64_2022-10-08.AppImage/file

If it crashes, please open a terminal, `cd` to the AppImage directory, run the following and then paste the backtrace in this issue:

```sh
cd ~/Downloads # if your AppImage is in Downloads
echo -e 'run\nbt\nquit' | gdb ./BambuStudio_ubu64_2022-10-08.AppImage
```

wms, 2022-10-08:
@deftdawg You're doing great work with these PRs and suggestions, really appreciate the effort you're putting in. Thanks.

BarrMo, 2022-10-09:
> Lastest AppImage built from my branch is here:
> 
> https://www.mediafire.com/file/4dbb1k21wsinhcg/BambuStudio_ubu64_2022-10-08.AppImage/file
> 
> If it crashes, please open a terminal, `cd` to the AppImage directory, run the following and then paste the backtrace in this issue:
> 
> ```shell
> cd ~/Downloads # if your AppImage is in Downloads
> echo -e 'run\nbt\nquit' | gdb ./BambuStudio_ubu64_2022-10-08.AppImage
> ```

Great! This works for me. Thank you SOOO much!

johnmeacham, 2022-10-09:
The AppImage is crashing for me on an assertion failure. ubuntu 20.04. 

```
(bambu-studio:35130): Gtk-CRITICAL **: 02:10:43.391: gtk_window_resize: assertion 'width > 0' failed
bambu-studio: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

Thread 1 "bambustu_main" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff1458859 in __GI_abort () at abort.c:79
#2  0x00007ffff14cb2da in __malloc_assert
    (assertion=assertion@entry=0x7ffff15ef8a8 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", file=file@entry=0x7ffff15eb3d6 "malloc.c", line=line@entry=2379, function=function@entry=0x7ffff15f0030 <__PRETTY_FUNCTION__.13066> "sysmalloc") at malloc.c:298
#3  0x00007ffff14cd93f in sysmalloc (nb=nb@entry=3088, av=av@entry=0x7ffff1622b80 <main_arena>) at malloc.c:2379
#4  0x00007ffff14ce793 in _int_malloc (av=av@entry=0x7ffff1622b80 <main_arena>, bytes=bytes@entry=3072) at malloc.c:4141
#5  0x00007ffff14d0299 in __GI___libc_malloc (bytes=3072) at malloc.c:3066
#6  0x00005555584fb167 in wxImage::Create(int, int, bool) ()
#7  0x00005555562f26f8 in Slic3r::GUI::BitmapCache::insert_raw_rgba(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, unsigned char const*, bool) ()
#8  0x00005555562f410f in Slic3r::GUI::BitmapCache::load_svg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float) ()
#9  0x000055555664f484 in create_scaled_bitmap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, wxWindow*, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) ()
#10 0x00005555566500aa in ScalableBitmap::ScalableBitmap(wxWindow*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, bool, bool)
    ()
#11 0x00005555565b0b83 in Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
    ()
#12 0x00005555565b2255 in Slic3r::GUI::TabPrint::build() ()
#13 0x00005555565ae4bc in Slic3r::GUI::Tab::create_preset_tab() ()
#14 0x000055555647b575 in Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    ()
#15 0x000055555647c83a in Slic3r::GUI::MainFrame::create_preset_tabs() ()
#16 0x0000555556489219 in Slic3r::GUI::MainFrame::init_tabpanel() ()
#17 0x0000555556489dab in Slic3r::GUI::MainFrame::MainFrame() ()
#18 0x000055555644edcd in Slic3r::GUI::GUI_App::on_init_inner() ()
#19 0x000055555644ffca in Slic3r::GUI::GUI_App::OnInit() ()
#20 0x00005555587ad377 in wxEntry(int&, wchar_t**) ()
#21 0x0000555556426e8b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#22 0x0000555555d1834b in Slic3r::CLI::run(int, char**) ()
#23 0x0000555555c6fa59 in main ()
```

deftdawg, 2022-10-10:
@probonopd I had a go at getting the dependencies into the AppImage because SteamOS (on SteamDeck) doesn't provide everything by default...  

In particular WebKit is missing, however this is a bit of an issue because the Ubuntu distro's `libwebkit2gtk-4.0.so.37` hardcodes the path to it's `WebKitNetworkProcess` executable under `/usr/lib/x86_64-linux-gnu/webkit2gtk-4.0`, this path needs to come out of `$AppDir/usr/bin//webkit2gtk-4.0`, because `/usr/lib/x86_64-linux-gnu/webkit2gtk-4.0` doesn't exist in SteamOS by default... 

Short of requiring root to bind mount it, if it doesn't exist, I'm not sure what I can really do about that... 

```
root@849e8fddabe9:/BambuStudio/build/package/usr/bin# strings libwebkit2gtk-4.0.so.37  | grep x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/webkit2gtk-4.0
/usr/lib/x86_64-linux-gnu/webkit2gtk-4.0/injected-bundle/
/usr/lib/x86_64-linux-gnu
root@849e8fddabe9:/Bamb
``` 

> Are these dependencies really needed? If so, they should probably be deployed into the AppImage because they cannot be expected to be "there" on all Linux distributions by default.
>
> ``` [...]

[312 more replies]