bambulab/BambuStudio #9: Slicer won't slice even after matching temperatures of dual material print (temperature match error)

kvnper, 2022-07-16:
**OS:**
Windows 10 64-bit

**Issue:**
The slicer won't slice due to the following error:
[image]
"Can not print multiple filaments which have large difference of temperature together. Otherwise, the extruder and nozzle may be blocked or damaged during printing"

But I have matched the temperatures of both filaments, and the error won't go away:

(Note: I am not actually going to print PVA at this temperature, I am just testing the slicer.)

[image]

[image]

[image]

ChrisG-hub, 2022-07-16:
To be honest, print temperature of PVA is between 190°C en 220°C where ABS needs higher temps. Maybe you should considering  HIPS Filament, that is the one i use for ABS/ASA.
Not to be rude, but this question could be posted on discord or FB and is in no way an question for the developers.

kvnper, 2022-07-16:
> To be honest, print temperature of PVA is between 190°C en 220°C where ABS needs higher temps. Maybe you should considering HIPS Filament, that is the one i use for ABS/ASA. Not to be rude, but this question could be posted on discord or FB and is in no way a question for the developers. So, ABS/PVA is not a match and yes they require different temps to be printed. HIPS is the one to use for ABS. HIPS and ABS have similar strengths & stiffnesses and print at around the same temperature (approx. 230°C). As a result, these two filament types compliment each other well, particularly when using dual extruders during the same printing process.
> 
> PLA has a significantly lower printing temperature than HIPS, making the two materials incompatible during the same print process. As a result polyvinyl alcohol (PVA) would be a better alternative as a support material as it and PLA have similar printing temperatures. PVA is suited for the lower temps like PLA

I don't mean to be rude either but if you actually read my post you would understand this is a slicer bug. Opinions on what material which should be used is an entirely different matter.

ChrisG-hub, 2022-07-16:
You are not rude, and i am not offended.
If the PVA toption is hardcoded 'temp'wise it would be normal to raise a warning. 
I did not know there are high-temp PVA filaments, technology evolve, the fact that HIPS is so cheap wont change my mind on trying with PVA high-temp filaments.

Back on topic, considering this is still a beta version, maybe they should lock the option to raise temp or state somewhere that there is a temp-lock for that material.

kvnper, 2022-07-16:
Perhaps we should wait on an official response.

As I've screenshotted above, the temps can be changed in the GUI.

whutwei, 2022-07-26:
Based on various existing materials, we have matched the materials that can be mixed and printed according to their optimal printing temperature range, in order to avoid the blockage caused by the temperature difference of different materials. For example, ABS and ASA, PLA and Support W can be mixed.

    The reason why we restrict the material rather than the temperature is as follows: if the user prints two materials with a big difference in optimal temperature, such as PLA and ABS, both use the temperature of 250, one of the materials (PLA) will have the risk of plugging if used at high temperature for a long time.

To minimize that risk, we adopted this strategy

kvnper, 2022-07-26:
I see.

What do you recommend for the user to do if using the following PVA
filament, which can be used up to 250c temperature?

https://www.formfutura.com/shop/product/helios-support-2814?category=461

SaltWei, 2022-08-09:
Has removed the limitation that PVA can't be sliced with ABS together

YifanWuBambu, 2022-08-11:
@kvnper 
Please update to v1.1.1.1.

closed 2022-08-11 by YifanWuBambu

addictgamer, 2023-11-30:
> Has removed the limitation that PVA can't be sliced with ABS together

> kvnper Please update to v1.1.1.1.

Hmm, I'm on v1.7.7.89 and seeing this even though my PVA and ABS filaments' ranges overlap. Was this reverted?
If I change the PVA's type to say ABS (without modifying the temperature) the error goes away.

kvnper, 2023-11-30:
> filaments' ranges overlap. Was this change reverted?

It's been working for me ever since, up to the same version. It's been a
long time since I set up the filament profile. You may need to turn on
developer options and then go into the filament settings and tick the
"support settings" checkbox.