bambulab/BambuStudio #18: Unable to "detect thin walls" when wall continuity is broken.

thrutheframe, 2022-07-17:
Version: Pre-release Windows 01.00.10.05

Settings: 
Printer - Bambu Lab X1 Carbon 0.4 nozzle
Process - 0.20mm Standard @BBL X1C
Walls - Wall loops =3 and "Detect thin walls" enabled

Issue: when slicing an object with non-continuous thin wall under 2mm. it will produce a single wall with infill. (see image). Both enabling and disabling "Detect thin walls" yield the same results. 

[image]

Cosmo991119, 2022-07-19:
Could you provide your model of this issue?

thrutheframe, 2022-07-19:
[wall_test.zip](https://github.com/bambulab/BambuStudio/files/9137330/wall_test.zip)
@Cosmo991119 attached in the zip file is the the stl and 3mf file. It occur when I use 3 wall loops.

whutwei, 2022-07-22:
Wait next release and try again. Thanks

thrutheframe, 2022-07-23:
Issue resolved with the recent update of Bambu Lab Slicer v01.01.00.13. Thanks

The option for "Detect thin wall" is also removed.

Will close this issue. 

[image]

closed 2022-07-23 by thrutheframe