bambulab/BambuStudio #19: Release

DanielDimmick, 2022-07-17:
Using https://github.com/bambulab/BambuStudio/releases/tag/v1.0.10 on fresh Windows 11 application does not start with the following 2 events:

```
Faulting application name: bambu-studio.exe, version: 1.0.0.0, time stamp: 0x62d2630c
Faulting module name: BambuStudio.dll, version: 0.0.0.0, time stamp: 0x62d2630a
Exception code: 0xc0000005
Fault offset: 0x0000000000a4c6e0
Faulting process ID: 0x4734
Faulting application start time: 0x01d899e771e99b97
Faulting application path: C:\Program Files\Bambu Studio\bambu-studio.exe
Faulting module path: C:\Program Files\Bambu Studio\BambuStudio.dll
Report ID: ede77d47-21f1-4fca-959b-e332e5df4dc0
Faulting package full name: 
Faulting package-relative application ID: 
```

and

```
Faulting application name: bambu-studio.exe, version: 1.0.0.0, time stamp: 0x62d2630c
Faulting module name: BambuStudio.dll, version: 0.0.0.0, time stamp: 0x62d2630a
Exception code: 0xc000041d
Fault offset: 0x0000000000a4c6e0
Faulting process ID: 0x4734
Faulting application start time: 0x01d899e771e99b97
Faulting application path: C:\Program Files\Bambu Studio\bambu-studio.exe
Faulting module path: C:\Program Files\Bambu Studio\BambuStudio.dll
Report ID: f4116800-318d-4301-bdc6-7519af951acc
Faulting package full name: 
Faulting package-relative application ID: 
```

wesleypark, 2022-07-17:
Same error codes on Windows 8.1

jwalker55, 2022-07-17:
Same issue, Windows 10 21H2
i7-9700k
32 GB
RTX 3070 Ti

lanewei120, 2022-07-18:
The problem may be caused by Windows Media Player not being installed,
Could you install the media player from this page [window media player]( https://support.microsoft.com/en-au/windows/get-windows-media-player-81718e0d-cfce-25b1-aee3-94596b658287) and have a try again?

Meanwhile we will fix the crash issue even it is not installed in the next release soon

wesleypark, 2022-07-19:
Yup. Looks like you need WMP. Other media players don't work.

DanielDimmick, 2022-07-24:
I will close this issue as it does resolve it however 'windows media player' is deprecated in favour of 'media player' and should not be required.

closed 2022-07-24 by DanielDimmick