bambulab/BambuStudio #20: The view should not be changed between Prepare mode and Preview mode, just as it doesn't change from Preview to Prepare mode

LarryVand, 2022-07-17:
The vast majority of times you are looking at a specific area in Prepare mode and you want to see how it will look when sliced. By changing the perspective view, as currently being done in Preview mode, you are forced to again rotate, pan, zoom, the sliced model to see the same exact area. So if you are going to leave it the same when you go back from preview mode to prepare mode, then you should leave it the same the other way too. It is visually disturbing for no reason. And IMO the unique visual clues of the BL plate on the screen (plate extension at the front and the different nozzle wipe extension at the back) tell you exactly how the print is oriented. So there is really no reason to change the view from one mode to the other.

BTW, prusaslicer and superslicer also do not change the view when moving from 3D editor mode to preview mode and then back again.

wuyifanfancy, 2022-07-22:
This is in our plan. We will optimize the view when changing between Prepare and Preview modes.

Gio76, 2022-08-09:
Issue fixed in the latest release (v.1.1.1.1)

closed 2022-08-10 by YifanWuBambu