bambulab/BambuStudio #21: [Feature Request] Enable UI Mode Switching (Light/Dark)

ptyork, 2022-07-17:
Please enable the ability to switch between Light and Dark modes. It appears much of the "plumbing" made the move from PrusaSlicer...at least as well as I can ascertain from perusing the code. If not an explicit GUI switch, then change colors based on the user's OS theme choice.

I know for most this is simply a nice-to-have. For some (myself sadly included), this is an accessibility issue. The tool is effectively unusable for me with a bright background. Unfortunately, I cannot even use Windows built-in tools to invert the screen with this slicer because half of the UI is already dark (build plate, window chrome, etc.) and thus it just flips which half is too bright.

wuyifanfancy, 2022-07-22:
This is in our plan, but maybe not in recent release.

TheCableGuy99, 2022-11-09:
This is a must please.

ptyork, 2022-12-15:
Thank you for pushing this into the 1.4 build! This issue can now be closed. Woo hoo!!

closed 2022-12-15 by ptyork