bambulab/BambuStudio #22: Search box remains on top of other applications when switching with Alt+Tab

RobinWhitfield, 2022-07-17:
When clicking the search box...
[image]

and then using Alt+Tab to change to a different application, the search box remains over the top of the other applications.
[image]

It won't disappear until you return to the application, or click it (which returns you to the application).

Windows 11
21H2
01.00.10.05
Installed from the pre-release Windows installer.

wuyifanfancy, 2022-07-22:
We will fix the bug in recent release.

RobinWhitfield, 2022-08-26:
Just noting that this issue is still present in 01.01.01.01

RobinWhitfield, 2022-09-09:
Still present in 01.02.00.04

RobinWhitfield, 2022-09-13:
Still present in 01.02.00.09. despite being marked as a fixed bug

[image]

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu