bambulab/BambuStudio #23: [Bug] Hitting the slice button adjusts the current user view to Top view

LarryVand, 2022-07-17:
You are in Preview mode and you pan, zoom, and rotate in a specific area to see your supports how they are sliced. You adjust your support settings and re-slice only to see  the view change again. The view should not change every time you hit the Slice button. And should not change every time you go to Preview mode either. The xyz current user view (Perspective or Orthogonal) should stay the same thru different modes or sliced. It is very visually disturbing and disorienting to the user, not to mention unproductive, in having to adjust the view every time they switch to the Preview mode or every time they slice.

wuyifanfancy, 2022-07-22:
Thanks for the suggestion. We will optimize the view when switching between Prepare and Preview page, in both single and multiple plates senarios.

YifanWuBambu, 2022-08-09:
@LarryVand The optimization is in the latest version. Please download and try it.

closed 2022-08-09 by YifanWuBambu