bambulab/BambuStudio #24: [Bug] Print>Send print job to dialog is not centered on screen

LarryVand, 2022-07-18:
Print>Send print job to dialog is not centered on screen. Send button is not shown till dialog is dragged and moved higher.

RobinWhitfield, 2022-07-18:
Presuming this is on Windows, what resolution/scaling are you using?

Can you attach a screenshot?

It's pretty centred on a 1080p screen at 100% scaling on Windows 11:
[image]

LarryVand, 2022-07-18:
Unfortunately it is not centered on a 1920x1080 at 125% scaling on Window 10 Home Version 10.0.19044
[image]

RobinWhitfield, 2022-07-18:
As they seem to be hardcoding the application size maximum to your primary monitor (#7), perhaps they're also using that resolution for pop-up window positioning, ignoring scaling.

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu