bambulab/BambuStudio #25: slow scroll key search.

MarcoPiazza83, 2022-07-18:
When the search button is clicked, the drop-down menu opens.
Scrolling is slow; the sidebar must be used to get to the bottom of the menu.
[image]

RobinWhitfield, 2022-07-18:
Scrolling speed down this list seems OK here. What input device are you using, and how many lines at a time is your computer set to scroll? The default on Windows I believe is 3, which is what mine is set to.

[image]

MarcoPiazza83, 2022-07-18:
> La velocità di scorrimento verso il basso di questo elenco sembra OK qui. Quale dispositivo di input stai utilizzando e quante righe alla volta è impostato per lo scorrimento del tuo computer? L'impostazione predefinita su Windows credo sia 3, che è ciò su cui è impostato il mio.
> 
> [image]

3

[image]

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu