bambulab/BambuStudio #26: View all object's setting ( version 01.00.10.05/ Win )

MarcoPiazza83, 2022-07-18:
the graphic setting of this screen is not optimized.
Long names are cut off, the box is not self-adaptive.
Sometimes it takes one click, sometimes two to select rows.

[image]

IBH7493, 2022-07-18:
Very nice work

MarcoPiazza83, 2022-07-18:
Imac 27", parallels desktop, win 10

[image]

> What resolution and scaling are you using? I suspect this is related to #24
> 
> On 1080p, 100% scaling it looks OK, [image]
> 
> It _does_ however let you resize the window which isn't very helpful (it doesn't give you a scroll bar) [image]
> 
> And then when you close and re-open it it breaks in weird and wonderful ways [image]
> 
> No, you cannot resize that left hand side section, but at least you get a scroll bar at the bottom now! Thankfully this does not persist with a software restart.

RobinWhitfield, 2022-07-18:
I opened #30 for the related issue that I found, for better visibility, as it's clearly a different issue.

wuyifanfancy, 2022-07-22:
The part list/settings panel has not been fully developed yet. We will consider your suggestions in the subsequent development process.

lanewei120, 2022-09-09:
the object table has been released formally in v1.2.0
please have a try and verify

RobinWhitfield, 2022-10-11:
@MarcoPiazza83, as #24 has been closed, is this one also resolved for you now?

YifanWuBambu, 2022-12-03:
@MarcoPiazza83 
We close it for now. Please create another issue if you still have this problem.

closed 2022-12-03 by YifanWuBambu