bambulab/BambuStudio #28: [Bug / 01.00.10.05 / Win ] Forced closure simulation

MarcoPiazza83, 2022-07-18:
Simulating a forced shutdown, on restart you are asked whether to load the last save. 

[image]

The loaded stl file loses triangles in the mesh.

Original

[image]

After forced shutdown

[image]

RobinWhitfield, 2022-07-18:
Where the numbers exist on those parts on the right, do they actually have depth? If so, how much depth is present?

Testing with the following object in both STEP, STL, and multiples of the STL I can't get it to lose any triangles.

[image]

[image]

[image]

[image]

[image]

[image]

MarcoPiazza83, 2022-07-18:
@RobinWhitfield 

Right click-> reduce triangles.
Same.

did you force close from task manager?

RobinWhitfield, 2022-07-18:
Yeah if I right click and reduce triangles then I get exactly the same triangle count as the box bottom right says.

I was force closing with task manager, and getting the same unsaved project message.

I'm presuming it's something specific with your model, as it's the numbers that are disappearing, and they look to have either a very small depth, or no depth at all.

If you slice the original file, does it do anything with the numbers, or is it trying to ignore them thinking they're not relevant?

MarcoPiazza83, 2022-07-18:
@RobinWhitfield 

Could you please share your stl file?
So I perform more tests.
I had the same error with other stl as well.
The file in screenshot is a thingiverse download.

RobinWhitfield, 2022-07-21:
It's just a random file I generated for testing.

I've just tested it with a load of Benchy's grouped as an assembly and it's seeming OK.
The exact model being the .STL from here: https://www.thingiverse.com/thing:763622/files

Before
[image]

After
[image]

If you want to share the Thingiverse download for what you've been using then I'll see if I can replicate the issue with that model too.

Maybe it's an issue with multi-part STLs?

wuyifanfancy, 2022-07-22:
This bug has been fixed. It will be contained in the next release.

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu