bambulab/BambuStudio #30: [Bug, UI / Win] Object/part setting can be resized, without scroll bar. Reopening does not correctly revert.

RobinWhitfield, 2022-07-18:
You can resize the Object/Part window, but no scroll bar appears when you do.
[image]

After resizing, if you close and re-open it, the content seems to remember your last size in some manner.
[image]

Restarting the software reverts the content size.

01.00.10.05
Windows 11
21H2

wuyifanfancy, 2022-07-22:
The Object/Part table has not been developed yet. We will fix the  in the subsequent development process.

SaltWei, 2023-04-21:
[image]
should be OK one latest 1.6 public beta

closed 2023-04-21 by SaltWei