bambulab/BambuStudio #32: [REQUEST] Add pause gcode after calibrations

alexsanchiz, 2022-07-18:
I would like if you could add a check box which injects a pause gcode after calibrations and just before printing, so we can remove calibration prints and print a full flat plate.

tsdexter, 2022-08-15:
@wuyifanfancy alternatively, could we move the calibration pattern elsewhere so we can print a large diameter circle with the calibration in the center of the build plate to avoid having to pause and restart the print?

YifanWuBambu, 2022-09-13:
It is added in v1.2.

closed 2022-09-13 by YifanWuBambu

tsdexter, 2022-12-12:
where is this setting located @YifanWuBambu ?

YifanWuBambu, 2022-12-13:
@tsdexter 
1. slice and model and go to preview page
2. drag the vertical slider bar to the bottom
[image]
3. insert Pause in the context menu