bambulab/BambuStudio #35: Inconsistent behaviour with support on bridges

chopsywa, 2022-07-20:
It seems that the "Max bridge length only works sometimes.

If I start a new model, set to 0.24mm Draft @BBL X1C and turn on support, I can load a model and supports will be spaced 20mm apart on bridges (20mm is the default value).

If I then change to
 any other quality level (lower, or higher) I get solid supports across a bridge. Further I cannot then switch back to .24mm. I have to restart the program to get it to work on my bridge.
[image]
[image]

tangketan, 2022-07-22:
Hi! We'll release a new version soon, in which the behavior of "Max bridge length" is redesigned. This bug could not be repeated in this new version. Try out! And please let us know for any further issue.

chopsywa, 2022-07-23:
I have installed 10.1.0.13 and the behavior is consistent now. Thanks.

closed 2022-07-26 by wuyifanfancy