bambulab/BambuStudio #41: Feature request: Right click option

ChrisG-hub, 2022-07-22:
Feature request: Right click option in the sliced window to add a 'gcode' in the model, similar to the 'pause' option witch is available now.

wuyifanfancy, 2022-07-22:
Hi @ChrisG-hub ,
Can you describes the specific senario in which you use this feature?

ChrisG-hub, 2022-07-22:
Sure,Sure,
This is an option in Prusaslicer in sliced view where you can add this function similar to the option already in Bambu slicer 'add pause'.
This is useful for adding gcode that could change temp or other gcode functions.
Here is the screenshot in Prusaslicer where you can find that function.
[image]

wuyifanfancy, 2022-07-25:
Thanks for the response. We will consider this feature in future release.

th3fallen, 2022-08-29:
+1 on this would be realllllly useful

YifanWuBambu, 2022-08-31:
@ChrisG-hub @th3fallen  We will add this feature in recent release

YifanWuBambu, 2022-10-21:
Please update to v1.3.0.12.

closed 2022-10-21 by YifanWuBambu