bambulab/BambuStudio #44: Feature Request - Fuzzy Skin Tuning

Jquig76, 2022-07-22:
I see that we have the option to turn on fuzzy skin but it does not appear that when doing so we have an opportunity to fine tune exactly how we would like that fuzzy skin to look. Currently that is available in both Prusaslicer and Cura. Will that be an option in Bambu Studio?

SaltWei, 2022-10-26:
Good point!
Please wait the next release. We will provide same more settings like prusa for user to tune.

Kraminator, 2022-11-06:
Can you please also add that if the top surface is straight, there is an option to have that fuzzy too? Like if you have a box, you can only make the side walls fuzzy; the top can not be made fuzzy unless you make it rounded.

SaltWei, 2022-12-16:
> Can you please also add that if the top surface is straight, there is an option to have that fuzzy too? Like if you have a box, you can only make the side walls fuzzy; the top can not be made fuzzy unless you make it rounded.

Printer can't run fuzzy path in z axis, and top surface is always at same height, so this is impossible