bambulab/BambuStudio #45: Ugly UI font

tokamac, 2022-07-22:
Please fix the ugly Times font which permeates the whole interface of BambuStudio! 
This gives a very unprofessional look and feel to your app. 
We are not in the 1980s anymore and using such Serif typeset nowadays (in parts of the world using an alphabet and not ideographic logograms of course) really feels something went wrong and the software rolled back to some kind of default minimal configuration. Exactly like when a web site is broken by the accidental deletion of its main stylesheet on the server. 

Use Helvetica or even Arial if you wish, or better: adapt to the default system UI font provided by the operating system BambuStudio runs on. All serious apps do this.

Keep up the good work!

wuyifanfancy, 2022-07-25:
Thanks for the suggestion. We will optimize the UI for macOS recently.

brandonraineri, 2022-07-25:
+1 to this.  Seems like some windows use different fonts and it drives me crazy.  Their store has this too and IMO looks much better as a single font type.

YifanWuBambu, 2022-08-09:
It is Optimized in the latest version.

closed 2022-08-09 by YifanWuBambu

tokamac, 2022-08-09:
Wow. So fast. I am impressed by your reactivity!