bambulab/BambuStudio #47: macOS UI Optimisation

Gio76, 2022-07-23:
**BambuStudio version:** Version 01.01.00.13

**Operating system and version:** macOS Monterey 12.5

**Describe the bug:**
The macOS version of BambuStudio does not seem to be following the Apple Guidelines for window management. Apps for macOS should have the 3 buttons to minimise the window and most importantly to toggle the full-screen mode (see below example from another app, fully optimised for macOS):

- [image]

- Furthermore, the Menu of the application should reside in the top bar like this:
- [image]

- and not like this:
- [image]

- Some of the icons need to be adjusted since they are in very low resolution and on Retina displays this becomes evident when compared to high res icons.
- [image]

jpasqua, 2022-07-23:
In general the UI looks as though it is using a cross-platform porting layer rather than macOS-native controls. This should be rectified over time. It ought to be as native as PrusaSlicer/SuperSlicer.

RobinWhitfield, 2022-07-23:
Should be merged with #46

wuyifanfancy, 2022-07-25:
Thanks for the suggestions. We will optimize the UI for macOS recently.

Gio76, 2022-08-09:
Issues fixed in v1.1.1.1. Thank you for the excellent work @wuyifanfancy

closed 2022-08-09 by Gio76