bambulab/BambuStudio #48: Incompatible 3mf

aquatacizem, 2022-07-23:
Importing a 3mf file (I have tested files exported from Fusion 360 and also PrusaSlicer) causes an error - 'Bambu Studio - Incompatible 3mf', with the text "The 3mf is not compatible, loading geometry only!". In the case of a multi-part object, on clicking OK, the geometry loads in pieces piled together.

deruth, 2022-07-23:
This happens to me also.

RobinWhitfield, 2022-07-23:
I wonder whether the 3MF files from all of these applications (PrusaSlicer, Fusion360, Bambu Studio) conform to the standards correctly.

There's software you can run them through to check.
https://github.com/3MFConsortium/test_suites/releases/tag/v1.0.0

RobinWhitfield, 2022-07-23:
It seems that Bambu Studio complains about every single 3MF file I give it from the test models.

Perhaps the error should just be a notification to inform you that it wasn't created in Bambu Studio?

Importing a test file with multiple parts I get the same as @aquatacizem - all objects are imported and moved to touch the build plate.
[image]
[image]

SuperSlicer gives the option to import as a multi-part singular object:
[image]
[image]

aquatacizem, 2022-07-23:
@RobinWhitfield thanks for verifying this.
I do a lot of multi colour printing with the Prusa MMU2S unit, and my workflow involves exporting multi-body components built in Fusion 360 exported to 3mf format. PrusaSlicer can read the individual object names from the 3mf file, allowing for quick assignment of extruder colour for each object.

At the moment, in order to work with a multi-part file in BambuLab Studio I need to export as obj or stl, then use the 'split to parts' function in the slicer, and then go through each object one-by-one and assign the filament colour manually. Not a huge deal but can be more tedious and time consuming than necessary.

Overall, I really like the slicer so far and there is much promise and potential for a great, feature rich program here. So, thank you very much BambuLab team for your hard work.

candre23, 2022-07-24:
This is a simple 40mm cube created in MS 3d builder.  Dragging it into bambu studio generates the "this 3mf is not compatible" error.  Since bambu studio seems to generate this error with nearly every 3mf created with almost any software, and since it is perfectly capable of slicing the model anyway, clearly the error message is erroneous.  

[cube.zip](https://github.com/bambulab/BambuStudio/files/9175034/cube.zip)

RobinWhitfield, 2022-07-24:
The software is almost certainly checking for certain files within the 3MF that are created when you save in Bambu Slicer.

I think there are two parts to this. One being that the message should probably read "The 3MF file was not created with Bambu Slicer, loading geometry data only" (which is correct and perfectly descriptive), and the second being that it's not importing multi-part objects correctly.

YifanWuBambu, 2022-08-09:
Optimized in the latest version.

closed 2022-08-09 by YifanWuBambu