bambulab/BambuStudio #49: [Mac v01.01.00.13] App will crash when changing filament on object/part setting dialog

gregconnors, 2022-07-23:
Steps to reproduce:
1. In prepare mode
2. Select object
3. Ctrl/right-click on object and choose ‘Per object edit’ menu item - this will display dialog box/window with objects (note this is not the object list on the left edge)
4. Change filament
    1. Click once to select filament
    2. Click again for menu
    3. Click a filament from menu and app will crash

Attached Mac crash report
[Problem report.txt](https://github.com/bambulab/BambuStudio/files/9172978/Problem.report.txt)

wuyifanfancy, 2022-07-23:
Thanks for reporting this issue. This Part/Object panel is still underdevelopment and is hidden in the formal release.
We will fix the issue in recent release.

lanewei120, 2022-08-09:
this issue is fixed in v1.1.1.1

closed 2022-08-09 by lanewei120