bambulab/BambuStudio #50: [Request / 01.01.00.14 / Win] Enabling Chamber Fan clean air through the air filter

thrutheframe, 2022-07-23:
There is no access to control the camber fan in the slicer or printer's UI. I notice that the fan do not turn on while printing PLA. 

While printing PLA is relative safe. Printing fast and with high temperature will create VOC. It will be good for users to be able to enable the chamber fan to pull the VOC over the active carbon filter. It can also be used to control the chamber temperature via the chamber fan.

RobinWhitfield, 2022-07-23:
As I understand it, it's already used to automatically control the maximum chamber temperature, and automatically controlling lower chamber temperatures is in the works, so giving users manual control over it would break this functionality.

If the chamber is relatively well sealed then presumably the best course of action is to (while printing) contain the fumes within the printer enclosure, before exhausting them at the end of the print.

SaltWei, 2023-02-22:
Already have access to control the camber fan in the slicer or printer's UI

closed 2023-02-22 by SaltWei