bambulab/BambuStudio #54: Feature Request - Add Arachne slicing function.

RobinWhitfield, 2022-07-23:
As per the title.

PrusaSlicer 2.5 Alpha now has the Arachne slicing function - it would be great to have this implemented in Bambu Studio soon too.

wuyifanfancy, 2022-07-23:
Will add this feature in recent release.

moebis, 2022-08-13:
@wuyifanfancy when you say "recent release" do you mean 01.01.01.01 version released a couple of days ago has Arachne now? I didn't see it in the release notes.

psiberfunk, 2022-08-22:
SuperSlicer also now adding this .

hutje, 2022-08-27:
Would be a really valuable feature for printing with 0.6mm nozzle when printing with filaments that would otherwise clog the 0.4mm nozzle.

danielclasen, 2022-08-29:
@hutje yeah, plus omitting those nasty gap fills between two walls that are just a hair apart the risk of a clog is even more reduced

psiberfunk, 2022-09-07:
Yea, the 3D printing world has quickly all decided that Arachne is pretty awesome with relatively few downsides.  Cura really knocked it out of the park with this feature.

RobinWhitfield, 2022-09-09:
This has been added, but I'm leaving the request open as it's only been added for specific slicing tasks.

I wholly expect additional Arachne in the future.

psiberfunk, 2022-09-11:
> This has been added, but I'm leaving the request open as it's only been added for specific slicing tasks.
> 
> I wholly expect additional Arachne in the future.

Where has this been added ?

RobinWhitfield, 2022-09-11:
You may have missed it, but 1.2.0.4 was released as a binary before they discovered a selection of bugs and removed it again.

I'm sure it'll be back again soon.

psiberfunk, 2022-09-11:
But to my other questions : how is this a partial implementation of Arachne?

RobinWhitfield, 2022-09-11:
I'm by no means an expert but I believe right now it's being used to avoid gap infill, but not perimeters. I think PrusaSlicer is almost the other way around, not using it for gap fill, but using it for perimeters

SaltWei, 2022-10-24:
Arachne has confliction with overhang slowing down and overhang cooling feature of BambuStudio, so we can't apply it for wall at the moment.
When arachne is applied to walls, the line width of wall may be various rather than fixed. And then it's difficult to calculate how much percentage of width of the wall extrusion line is unsupported.

We are still trying to fix this confliction.

psiberfunk, 2022-10-24:
You might look at Superslicer which has Arachne AND custom settings for overhangs

Lesani, 2022-11-30:
would be important for objects containing letters...
Sliced with Prusaslicer 2.5.0 (alpha3)
[image]

Sliced with Bambu Studio 01.03.01.01
[image]

Photogad, 2022-12-08:
Yesss,full Arachne! 🕷️

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu