bambulab/BambuStudio #56: [Bug] Trojan/Emotet.mv

gofromzero, 2022-07-23:
[image]
下载网络插件时，会报木马 这是什么情况

wuyifanfancy, 2022-07-25:
@gofromzero 
What's the security software are you using?
We will check the issue on the same security software.

gofromzero, 2022-07-25:
火绒安全 win   https://www.huorong.cn/

martinfrancois, 2022-07-26:
I also noticed on macOS, the "Web Shield" component in "Avast One" anti virus blocks connections to both download the network plug-in and do the login or connection to the printer. I tried adding exclusions for `bambulab.com` and `iotcplatform.com` but this didn't help, I need to turn off the "Web Shield" component in order to be able to use this feature.

YifanWuBambu, 2022-08-31:
It is the same issue at #149.
[image]

closed 2022-08-31 by YifanWuBambu