bambulab/BambuStudio #61: Feature Request - File send from studio save to the printer

winnainglin, 2022-07-23:
The files print from the studio should save to the printer's internal or SD card for the future print.

RobinWhitfield, 2022-07-24:
Ideally I think this would be an option when sending to print. "Save to SD card?" as a tick box.

Perhaps the slicer could also tell you if no SD card was present and that wasn't possible.

martinfrancois, 2022-07-26:
I think it should be possible to send the files to the printer and save them to the SD card without starting a print at the same time as well, since I like to prepare a few plates in advance and then print one after the other without having to open the slicer every time. Currently the only way to do this is by using an SD card, which is inconvenient.

dereknheiley, 2022-08-12:
> Ideally I think this would be an option when sending to print. "Save to SD card?" as a tick box.
> 
> Perhaps the slicer could also tell you if no SD card was present and that wasn't possible.

or just silently fail to save to missing sd card on printer

closed 2023-02-22 by SaltWei