bambulab/BambuStudio #62: [BUG / 01.01.00.14 / Win] Closing Bambu Studio not on the main screen makes the app unusable

Brennholz3000, 2022-07-24:
If I move Bambu Studio to my 2nd or 3rd screen and close it there, no matter maximised or not, it will crash every time i try to launch it again. To make it work again, I need to go to *user*/AppData/Roaming/BambuStudio and remove the line "window_mainframe" from the BambuStudio.conf file where the destination and size of the app window is stored. This makes the App start from its original position on the main screen again. If I close it, while its on my main(1st) screen, everything works fine. I tried to reinstall Bambu Studio, but the issue persists.

Some data that might be helpful:
Resolution 1st screen: 2560x1440@144hz - 100% scale
Resolution 2nd screen: 2560x1440@60hz (original resolution is 3840x2160 but I scaled it down) - 125% scale
Resolution 3rd screen: 1920x1080@60hz - 100% scale
I'm using the newest version of Bambu Studio but also had the issue in 01.01.00.13.
Windows and drivers are up to date

If you need any more data, let me know. I'll try my best to help out!

wuyifanfancy, 2022-08-04:
@Brennholz3000 Is this bug? We fixed it and will be in next release version.

[image]

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu