bambulab/BambuStudio #63: Cannot print from cloud servers using Mac BS v01.01.00.14

reify3d, 2022-07-24:
Problem: Sending print job from Mac version (x86_64) of BambuStudio v01.01.00.14 fails 100% of the time. Refer to attached GIF. The same file prints without issue on a Windows 10 VM on the same machine.

[image]

Expectation: Clicking on "send" should start the print.

What actually happened: BambuStudio complains of print failure.

Workarounds: None

Hardware: Intel iMac 2019 27"
Software: macOS Monterey 12.4 (21F79)
Machine: BL X1 Carbon + AMS
Machine firmware: 00.01.03.26
AMS firmware: 00.00.04.21

martinfrancois, 2022-07-26:
For me this works with a similar system, do you have any firewall / anti virus programs installed? Try disabling any web protection components you have installed to see if that helps.

reify3d, 2022-07-26:
> For me this works with a similar system, do you have any firewall / anti virus programs installed? Try disabling any web protection components you have installed to see if that helps.

Hello, I have disabled macOS firewall as stated previously and it did not resolve the issue. All my other networking progams work, even the Windows VM works, so I do not think it's a network issue on the operating system.

reify3d, 2022-07-27:
This fragment repeats many times in a log file at this path: ~/Library/Application Support/BambuStudio/log

`, size = 390
2022-07-26 13:05:09.764669[Thread 0x0000700005120000]:print_job: start request project id
2022-07-26 13:05:10.274963[Thread 0x00007000029e1000]:message topic:device/[mydeviceid]/report, payload={}
2022-07-26 13:05:10.584516[Thread 0x0000700005120000]:AccountManager::request_project_id, body={"message":"success","code":null,"error":null,"project_id":"952","model_id":"[mymodelid]","status":null,"name":null,"content":null,"create_time":null,"update_time":null,"profiles":null,"profile_id":"952","upload_url":"http://upload-file.bambulab.com/users/[mydeviceid]/models/USbeb11341ccb780/profiles/952/1658811910.3mf?x-oss-traffic-limit=25165824&OSSAccessKeyId=LTAI5tFANCnfNNPNSSLjG97f&Expires=1658898310&Signature=GrdpPgXZCKdtd%2FTcnFtXm7jgiZc%3D","upload_ticket":"uploader:[mydeviceid]:USbeb11341ccb780:952:1658811910"}
2022-07-26 13:05:10.584689[Thread 0x0000700005120000]:print_job: get project id = 952
2022-07-26 13:05:10.584696[Thread 0x0000700005120000]:print_job: get profile id = 952
2022-07-26 13:05:10.584712[Thread 0x0000700005120000]:print_job: start to uploading...
2022-07-26 13:05:10.585034[Thread 0x0000700005120000]:print_job: upload_3mf md5 = E4EB4B8FB3FD13E310FF8BEF0A5A730C
2022-07-26 13:05:10.587027[Thread 0x0000700005120000]:print_job: uploading is failed
2022-07-26 13:05:10.587041[Thread 0x0000700005120000]:bambu_network_start_print, start_print returned error:-23
2022-07-26 13:05:12.298290[Thread 0x00007000029e1000]:message topic:device/[mydeviceid]/report, payload={}
2022-07-26 13:05:13.857648[Thread 0x0000700004814000]:ssdp_mac: on_packet_received, packet = NOTIFY * HTTP/1.1
Host: 239.255.255.250:1990
Server: Buildroot/2018.02-rc3 UPnP/1.0 ssdpd/1.8
Location: [myinternalIPaddress]
NT: urn:bambulab-com:device:3dprinter:1
NTS: ssdp:alive
USN: [mydeviceserialnumber]
Cache-Control: max-age=1800
DevModel.bambu.com: 3DPrinter-X1-Carbon
DevName.bambu.com: [myprintername]
DevSignal.bambu.com: -62
DevConnect.bambu.com: cloud
DevBind.bambu.com: occupied`

StoneLiBambulab, 2022-07-27:
Failed to upload printing file to oss server.  Is it successfully sending a job though more retrys?

reify3d, 2022-07-27:
It only gets sent through if I use the Windows VM. If it's in the Mac native interface, it pretty much never goes through. I did not want to try too many times as it may be misconstrued as a denial of service attack.

docangle, 2022-08-06:
I am having the same issue on my M1 Mac Book Pro, but my M1 MacBook Air works fine

Devastortag, 2022-08-27:
I had the same problem on M1 MacBook Pro. Disabling iCloud Private Relay have worked for me.

reify3d, 2022-08-28:
> I had the same problem on M1 MacBook Pro. Disabling iCloud Private Relay have worked for me.

Well I'll be damned! That is the solution. But why would turning it on pose a problem specifically for BambuStudio, and not for all applications with networking? Hope nothing sinister is happening in the background.

docangle, 2022-08-28:
It is simply the fact that Bambu Studio is using your IP as an identifying piece of information when downloading/uploading and Private Relay blocks this bit of information.

StoneLiBambulab, 2022-10-14:
There are several known VPN software lead to network error on mac platform：
1. iCloud Private Relay；
2. Cisco AnyConnect;

disable or uninstall the VPN software works.

closed 2022-10-14 by YifanWuBambu