bambulab/BambuStudio #66: Feature request: Simplified picking of AMS slot when printing single-material

PaulTheSimon, 2022-07-24:
If I print a single-material thing where I want to use the material sat in the 4'th AMS slot, I would currently have to add 3 new materials to my project, and manually paint the object(s) in the material I want.

To streamline printing of single-material prints, why not allow a simple AMS slot override at the "send" screen?
[image]

Sirwill1968, 2022-07-24:
I'd totally love to see this feature too.

wuyifanfancy, 2022-07-25:
Bambu Studio already support this feature. But it also requires the firmware update, which will be released soon.

wuyifanfancy, 2022-07-26:
@Sirwill1968 
The printer firmware v1.0.0.0 has been released. Please update the firmware and use the feature.
[image]

closed 2022-07-26 by wuyifanfancy