bambulab/BambuStudio #67: Z hieght not working in slicer

PowerChaos, 2022-07-24:
hey all
this is just a smal bug report of the slicer
when you import a multi file and want to move a file in the Z plane ( above bed , Z offset of +30 for example ) then it fails

this makes it impossible to adjust the wrong offsets , or to use multi color that need to be adjusted/moved a little bit to match up

can you check this please
Thank you
Greetings from PowerChaos

wuyifanfancy, 2022-07-25:
@PowerChaos 
Please try these steps and check whether it solves the issue.
[image]

PowerChaos, 2022-07-25:
ok thank you
making a assembly of the print ( group ) then it works 
this was my test file ( 3MF ) and seems it was just differend objects instead a single object
but when imported to prusa slicer it is a assembly object . from there the confusion
[CraneTruckV5.zip](https://github.com/bambulab/BambuStudio/files/9177541/CraneTruckV5.zip)

Greetings from PowerChaos

wuyifanfancy, 2022-07-26:
Bambu Studio and PrusaSlicer use different 3mf structure.
So the 3mf created in PrusaSlicer is currently not supported in Bambu Studio.

We have added this compatibility task to our recent plan.

SaltWei, 2023-02-22:
[image]

closed 2023-02-22 by SaltWei