bambulab/BambuStudio #69: [BUG / 01.01.00.14 / macOS] Printable Area locked

Gio76, 2022-07-25:
The Z-Height (Printable Area) of the printer profiles is locked to 250mm:

[image]

Is there a specific reason this value is not editable? Also, why is it set to 250 and not to 256 (as per the advertised print area)?

RobinWhitfield, 2022-07-25:
It's locked so that you can't inadvertantly slice g-code for the printer that is higher than the machine will go. That's absolutely what I'd expect for an inbuilt machine profile - same as the max movement speeds.

Now why it's locked to 250 rather than the advertised 256mm is a very different question...

wuyifanfancy, 2022-07-26:
We reserved 6mm due to the following reason:
1. z-hop feature
2. prevent uncleaned bottom of chassis

We are considering whether to make this config editable.

Gio76, 2022-08-09:
Solved in the latest version (v1.1.1.1)

closed 2022-08-09 by Gio76