bambulab/BambuStudio #70: [BUG / 01.01.00.14 / macOS] Can not load the network plugin. Can not login

ptbsare, 2022-07-25:
Way to reproduce:
1.Open BambuStudio.
2.Auto install the network plugin.
3.Click Finish.
4.UI Still shows plugin not installed and cannot login.
5.Restart BambuStudio. Same result as 1.
[image]
[image]
[image]

lanewei120, 2022-07-25:
after installing finish, what are the files in the plugins directory?
[image]

you can delete all the files in this directory and retry again
if this issue still can not be solved
Could you send these two files and the log in /Users/your_username/Library/Application Support/BambuStudio/log to us?

StephanMa, 2022-07-25:
Let me hijack this issue.
MacOS
Cannot install network plugin

Logfiles says: │2022-07-25 16:52:07.694261[Thread 0x0000000177f73000]:[download_plugin]: check the plugin from https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.00.00

Unfortunately... there is only the win-plugin... no Mac

lanewei120, 2022-07-25:
> [plugins.zip](https://github.com/bambulab/BambuStudio/files/9182136/plugins.zip) [image]
> 
> This is the files in plugins folder. There is no log folder or file.
> 
> delete all the files in plugins folder did not resolve the issue.

the plugin has been installed successfully
it seems they can not work on your machine

if you open the "Console" on your computer, and do it again
is there any log in the console?
if yes, please help to send them to us

meanwhile, we will find a dual-core i5 machine with macos 11.6 to have atry

ptbsare, 2022-07-25:
```log
[2022-07-25 23:37:28.742898] [0x0000000114e06e00] [trace]   Initializing StaticPrintConfigs
loading network module at /Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib
after dlopen, network_module is 0x0
[2022-07-25 23:37:29.503007] [0x0000000114e06e00] [info]    select_technology: change the printer technology from  to 
[2022-07-25 23:37:29.524969] [0x0000000114e06e00] [info]    get_current_slice_result:m_current_plate 0, current_plate 0x7fd89291b000
[2022-07-25 23:37:29.535432] [0x0000000114e06e00] [info]    3dx mac handler starts
[2022-07-25 23:37:29.535774] [0x0000000114e06e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2022-07-25 23:37:29.535811] [0x0000000114e06e00] [info]    3dx drivers loaded: NO
[2022-07-25 23:37:29.660569] [0x0000000114e06e00] [info]    create_playerNetwork plugin not ready currently!
[2022-07-25 23:37:29.744943] [0x0000000114e06e00] [info]    wxMediaCtrl2::DoSetSize: width: 574, height: 398, maxHeight: 323
[2022-07-25 23:37:29.759812] [0x0000000114e06e00] [info]    ImageGrid::SetStatus: No printers.
[2022-07-25 23:37:29.775508] [0x0000000114e06e00] [info]    wxMediaCtrl2::DoSetSize: width: 590, height: 323, maxHeight: 332
[2022-07-25 23:37:29.777952] [0x0000000114e06e00] [info]    wxMediaCtrl2::DoSetSize: width: 574, height: 288, maxHeight: 323
[2022-07-25 23:37:29.794363] [0x0000000114e06e00] [info]    reset: current result id 0 
[2022-07-25 23:37:29.794711] [0x0000000114e06e00] [info]    set_shape:current position {0,0}, new position {0, 0}
[2022-07-25 23:37:29.794825] [0x0000000114e06e00] [info]    reset_size:before size: plate_width 0, plate_depth 0, plate_height 0
[2022-07-25 23:37:29.794868] [0x0000000114e06e00] [info]    reset_size:after size: plate_width 257, plate_depth 257, plate_height 250
[2022-07-25 23:37:29.794912] [0x0000000114e06e00] [info]    set_pos_and_size: plate_id 0, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2022-07-25 23:37:29.794960] [0x0000000114e06e00] [info]    set_pos_and_size: with_instance_move 0, after, origin {0,0,0}, plate_width 257, plate_depth 257, plate_height 250
[2022-07-25 23:37:29.795005] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2022-07-25 23:37:29.795047] [0x0000000114e06e00] [info]    set_pos_and_size: plate_id 1, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2022-07-25 23:37:29.795090] [0x0000000114e06e00] [info]    set_pos_and_size: with_instance_move 0, after, origin {308.4,0,0}, plate_width 257, plate_depth 257, plate_height 250
[2022-07-25 23:37:29.795125] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2022-07-25 23:37:29.795159] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2022-07-25 23:37:29.795192] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2022-07-25 23:37:29.832286] [0x0000000114e06e00] [info]    get_enable_print_status: m_print_select 1, enable= 0 
[2022-07-25 23:37:29.832364] [0x0000000114e06e00] [info]    get_enable_slice_status: m_slice_select 1, enable= 1 
[2022-07-25 23:37:29.832395] [0x0000000114e06e00] [info]    update_slice_print_status m_slice_select 1: can_slice= 1, can_print 1, enable_slice 1, enable_print 0 
[2022-07-25 23:37:29.834309] [0x0000000114e06e00] [info]    create /3D/Objects in "/var/folders/pq/hwg_ygwx4y5246_h3_g1xq640000gn/T/bamboo_model/Mon_Jul_25/23_37_29#7"
[2022-07-25 23:37:29.834743] [0x0000000114e06e00] [info]    create /Metadata in "/var/folders/pq/hwg_ygwx4y5246_h3_g1xq640000gn/T/bamboo_model/Mon_Jul_25/23_37_29#7"
[2022-07-25 23:37:29.834886] [0x0000000114e06e00] [info]    create /lock.txt in "/var/folders/pq/hwg_ygwx4y5246_h3_g1xq640000gn/T/bamboo_model/Mon_Jul_25/23_37_29#7"
[2022-07-25 23:37:29.835195] [0x0000000114e06e00] [info]    create missing sub_path/var/folders/pq/hwg_ygwx4y5246_h3_g1xq640000gn/T/bamboo_model/Mon_Jul_25/23_37_29#7//Auxiliaries
[2022-07-25 23:37:29.836142] [0x0000000114e06e00] [info]    load_current_preset: enter
[2022-07-25 23:37:29.837275] [0x0000000114e06e00] [info]    put_other_changes
[2022-07-25 23:37:29.839716] [0x0000000114e06e00] [info]    load_current_preset: exit
[2022-07-25 23:37:29.839751] [0x0000000114e06e00] [info]    load_current_preset: enter
[2022-07-25 23:37:29.840804] [0x0000000114e06e00] [info]    put_other_changes
[2022-07-25 23:37:29.842423] [0x0000000114e06e00] [info]    load_current_preset: exit
[2022-07-25 23:37:29.842457] [0x0000000114e06e00] [info]    load_current_preset: enter
[2022-07-25 23:37:29.842602] [0x0000000114e06e00] [info]    reset: current result id 4294967295 
[2022-07-25 23:37:29.842738] [0x0000000114e06e00] [info]    load_print_as_fff 621: print is step done, posSlice 0, posSupportMaterial 0, psGCodeExport 0
[2022-07-25 23:37:29.844389] [0x0000000114e06e00] [info]    put_other_changes
[2022-07-25 23:37:29.846858] [0x0000000114e06e00] [info]    load_current_preset: exit
[2022-07-25 23:37:29.851191] [0x0000000114e06e00] [info]    put_other_changes
[2022-07-25 23:37:29.858097] [0x0000000114e06e00] [info]    main frame firstly shown
[2022-07-25 23:37:29.921158] [0x0000000114e06e00] [info]    finished the gui app init
[... cut ...]
[2022-07-25 23:38:02.408072] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2022-07-25 23:38:02.413444] [0x0000000114e06e00] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2022-07-25 23:38:02.430411] [0x0000000114e06e00] [info]    reset: current result id 0 
[2022-07-25 23:38:02.433426] [0x0000000114e06e00] [info]    reset: current result id 4294967295 
[2022-07-25 23:38:02.433792] [0x0000000114e06e00] [info]    reset: current result id 0 

```

ptbsare, 2022-07-25:
I think the problem here is that
```log
[2022-07-25 23:37:32.859251] [0x0000000114e06e00] [info]    initialize_network_module, error, dlerror is dlopen(/Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib, 1): Symbol not found: __ZNKSt3__115basic_stringbufIcNS_11char_traitsIcEENS_9allocatorIcEEE3strEv
  Referenced from: /Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib (which was built for Mac OS X 12.0)
  Expected in: /usr/lib/libc++.1.dylib

after dlopen, network_module is 0x0
[2022-07-25 23:37:32.859332] [0x0000000114e06e00] [info]    initialize_network_module, can not Load Library for /Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib
[2022-07-25 23:37:32.859353] [0x0000000114e06e00] [info]    on_init_network: on_init_network, load dll failed
[2022-07-25 23:37:32.859368] [0x0000000114e06e00] [info]    on_init_network: on_init_network, need upload network module
[2022-07-25 23:37:32.859386] [0x0000000114e06e00] [info]    unload_network_module, network module 0x0
[2022-07-25 23:37:32.859403] [0x0000000114e06e00] [info]    on_init_network, unload_network_module, result = 0
[2022-07-25 23:37:32.859420] [0x0000000114e06e00] [info]    restart_networking exit, m_agent=0x0
```
Symbol not found
Maybe incompatible libc++ version.

lanewei120, 2022-07-26:
> I think the problem here is that
> 
> ```
> [2022-07-25 23:37:32.859251] [0x0000000114e06e00] [info]    initialize_network_module, error, dlerror is dlopen(/Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib, 1): Symbol not found: __ZNKSt3__115basic_stringbufIcNS_11char_traitsIcEENS_9allocatorIcEEE3strEv
>   Referenced from: /Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib (which was built for Mac OS X 12.0)
>   Expected in: /usr/lib/libc++.1.dylib
> 
> after dlopen, network_module is 0x0
> [2022-07-25 23:37:32.859332] [0x0000000114e06e00] [info]    initialize_network_module, can not Load Library for /Users/ptbsare/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib
> [2022-07-25 23:37:32.859353] [0x0000000114e06e00] [info]    on_init_network: on_init_network, load dll failed
> [2022-07-25 23:37:32.859368] [0x0000000114e06e00] [info]    on_init_network: on_init_network, need upload network module
> [2022-07-25 23:37:32.859386] [0x0000000114e06e00] [info]    unload_network_module, network module 0x0
> [2022-07-25 23:37:32.859403] [0x0000000114e06e00] [info]    on_init_network, unload_network_module, result = 0
> [2022-07-25 23:37:32.859420] [0x0000000114e06e00] [info]    restart_networking exit, m_agent=0x0
> ```
> 
> Symbol not found Maybe incompatible libc++ version.

it seems the networking plugins can not be loaded under MacOS 11.x
we are debugging on this issue, and hope to fix it soon

lanewei120, 2022-07-27:
we have fixed this issue,
please try again on your macOS

ptbsare, 2022-07-28:
Confirm Fixed.

closed 2022-07-28 by ptbsare

jecoffey, 2022-07-29:
Confirm Fixed. Thanks a lot!