bambulab/BambuStudio #72: Cant Open BambuStudio on Windows 10

ADmakes, 2022-07-25:
I used the installer and all I get is a splash screen saying "loading configuration", then white box , then it quits. Ive done several installs/uninstalls, restarts, and repaired the vcredist2019

lanewei120, 2022-07-25:
Hello

Could you help to package the folder C:\Users\your_user_name\AppData\Roaming\BambuStudio\log into a zip
and upload it here to help us debug?

and also , could you help to show your windows version and your local language?

thanks

ADmakes, 2022-07-25:
Ok here is the log file and screenshot of system info. Language is english

lanewei120, 2022-07-25:
I can not find your screenshot and log file,
there must be something missed

ADmakes, 2022-07-25:
[AppData_log.zip](https://github.com/bambulab/BambuStudio/files/9182236/AppData_log.zip)
[image]

lanewei120, 2022-07-25:
there is indeed a crash when init
however, we can not find the detailed stack in the formal release version

could you give an email here?
I will send your a package of the same version with pdb file(symbol table), 
then you can try again , 
and if it crashes, there will be a detailed stack which will be helpful

sorry for the inconvenience

ADmakes, 2022-07-25:
adi_orio@hotmail.com

ADmakes, 2022-07-25:
I have solved the problem!  
Windows media player needed activation. 
Although WMP was installed, it was not activated, and activating it was not obvious. 
In the SETTINGS>OPTIONAL FEATURES I had to click MORE WINDOWS FEATURES>Check the box for MEDIA FEATURES> then check WINDOWS MEDIA PLAYER, and it will install. Then Restart and Bambu Studio opens. Sharing this screenshot to help others. Thanks for your help
[image]

closed 2022-07-25 by ADmakes