bambulab/BambuStudio #74: Text pop ups error - not displaying in the right format

TheRhion, 2022-07-25:
The pop ups in the build area screen are not displaying correcting as the font seems to be messed up as seen on the pictures.
This is using Bambu Studio Windows version on Windows 11 in English . 

[image]
[image]
[image]
[image]

wuyifanfancy, 2022-07-26:
Hi @TheRhion , we tested this issue on several computers, but did not reproduce it.
Could you help provide the following information?

1. Display settings
[image]

2. Your Operating System language

TheRhion, 2022-07-26:
Hi thanks for this I have adjusted the settings on the display and its working now.
My recommended setting was at the scale of 300% so changed it to 250% and its working now.

[image]
[image]
[image]

closed 2022-08-31 by YifanWuBambu