bambulab/BambuStudio #77: Disable z-hop above specified height.

brandonraineri, 2022-07-26:
Instead of completely disabling z-hop so we can utilize the full 256^3 build area in the workaround discussed in the wiki, make it where it is only disabled above a user specified height (ex input: 255mm) so we can utilize this function for 99% of the print.

SaltWei, 2023-08-18:
@BambooBugFixer Please add this next version

BambooBugFixer, 2023-08-23:
All right.

BambooBugFixer, 2023-10-23:
This feature will be available in a future version pretty soon.

GuzziRaz, 2023-11-28:
Now that we have this implemented (even in stable), please bump the machine settings for Printable Height to 256 in all profiles. There's no reason not to - it just unnecessarily adds to uneducated "reddit truths" that "Bambu's can't use their advertised print volume".

closed 2024-01-15 by SaltWei