bambulab/BambuStudio #80: [Feature Request] Surface Mode

Panacaguy, 2022-07-28:
Please consider adding a feature like the "Surface Mode" in Cura (within the Special Modes section) or the "Single Shell Surface Mode" in Ideamaker (within the Layer tab). Many 3D printable airplane designs require this feature to properly slice with only a single outside wall and no top or bottom surface. It allows non-manifold geometry to be printed and differs from spiral mode in that it can print both inside and outside surfaces that are not one continuous line.

wuyifanfancy, 2022-07-28:
@Panacaguy 
Here is the feature.
[image]

closed 2022-07-28 by wuyifanfancy

bubnikv, 2022-07-28:
Spiral vase is not the same feature.

We would be greatful if BambuLab implements the Cura Surface mode into Bambu Studio. We would then happily merge it into PrusaSlicer. Thank you for considering.

Panacaguy, 2022-07-28:
Spiral mode is not the same as Surface Mode. My write-up in the feature
request briefly explains this. Please look into what Surface Mode really is
and re-open this feature request.

Thanks,
Andrew