bambulab/BambuStudio #81: [BUG / 01.01.00.14 / macOS + Windows] Auto Arrange ignores multiple selection of objects with printable property set to on/off

reify3d, 2022-07-29:
Description of problem:
When multiple objects with the "printable" property set to *OFF*, and the user clicks on "Auto Arrange", only 1 object is set. The converse is also true i.e. if multiple objects with the "printable" property is set to *ON*, only one object is set to the printable section in the list field.

[image]

What I expected:
Selecting multiple objects with the "printable" property checkbox set to off, should set them all to be non-printable when the "Auto Arrange" icon is clicked. Conversely, setting multiple objects with the "printable" property checkbox set to on, should do the opposite.

What actually happened:
Only 1 object instead of multiple objects has their "printable" property toggled when "Auto Arrange" icon is clicked.

What has been tried:
Reinstalling and rebooting did not resolve the issue.

Workarounds: Each object has to be individually selected., and then click on "Auto Arrange". This is extremely inefficient.

BambuStudio version: 01.01.00.14
Operating system: macOS Monterey and Windows 10

wuyifanfancy, 2022-08-02:
We will solve the issue in the next release.

CptHungarian, 2022-08-02:
As a workaround, you can select all objects with CTRL+A (or Edit -> Select All).

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu