bambulab/BambuStudio #83: Printing multiple layers between color changes to reduce waste?

PaulTheSimon, 2022-07-30:
I want to voice the idea of printing multiple layers between color change in order to reduce filament waste at finer resolutions.
At a resolution like 0.12mm, it should be possible to perform 2 or 3 layers with any given color before performing the color change. 
Is this something that's been considered and tested?

If possible, it would reduce filament waste massively at higher resolutions.

randyre, 2022-08-18:
Great idea.

psiberfunk, 2022-08-22:
This is a cool idea but it’s also very complex to implement properly without collisions , and causes lots of z-hop movement .

PaulTheSimon, 2022-08-22:
> This is a cool idea but it’s also very complex to implement properly without collisions , and causes lots of z-hop movement .

It causes a lot of z-hop, but saves a whole lot of filament changes that involve far more moving parts.

psiberfunk, 2022-08-22:
> > This is a cool idea but it’s also very complex to implement properly without collisions , and causes lots of z-hop movement .
> 
> It causes a lot of z-hop, but saves a whole lot of filament changes that involve far more moving parts.

Sure, I agree, I'm just saying there's a lot of extra considerations in implementing the feature.  Suddenly you've got to figure where and when to z-hop.. on every retract? only on retracts that would cross a part printed previously higher than the nozzle ?  How do you find all those previous higher paths and see if you're crossing them? etc.  It makes the implementation more complex.

rwoldberg, 2022-09-02:
I second this, especially when printing with a support interface layer and using expensive Support-W filament.  Seems like it should be possible to start printing with the currently loaded color if it is also on that layer.