bambulab/BambuStudio #87: Feature request ability to use more than 4 filaments with one AMS

CNCXPROUK, 2022-07-31:
If it’s possible it would be good to be able to program the g code to pause at a certain point and then swap filaments to another colour. For instance red to yellow then resume printing. In the slicer you could do this by selecting more than four colours and assigning the colour to a slot in the ams. The slicer would know when it needs to switch filament and pause the print waiting for the user. 

Although this may seem counter productive for selling extra AMS units, it wouldn’t. 

Those that want to buy extra AMS and want to walk away from the printer will be happy to do that.  Those that can’t afford that or don’t mind having to pause the print can do so.

Nico04, 2026-10-09:
Still relevant, and now on newer hardware. I have an X2D with an AMS 2 Pro + AMS HT (5 slots) and need a 6-color print: one color only at the start, another only at the end. They can share a slot, but Studio doesn't know a manual swap is needed. Today I map both filaments to the same slot and add a manual pause at the right layer by hand.
This is what this issue asks for: when more filaments than slots are used, the slicer should know and pause for the swap.
Related: #5361 (pause on filament change), #3469 (AMS + external spool, pause for manual intervention), #10261 (same gap with an external spool on a P2S).

DanaLeeTech, 2026-10-19:
I also ran into this problem, trying to print something with 9 filaments on my A1 with a single AMS. To make it easier for myself and others (until Bambu builds this), I made a tool which plans out the manual swaps and inserts automatic pauses mid layer to do the swaps - https://github.com/DanaLeeTech/PausePalette - only tested on my printer so far, but should work on others.

SaltWei, 2026-10-21:
> 5\. I also ran into this problem, trying to print something with 9 filaments on my A1 with a single AMS.

A1打印机已经支持了这个功能，请升级到最新的A1打印机固件，发送打印的时候在studio上勾选”外挂多色打印“的选项

DanaLeeTech, 2026-10-21:
> 6\. > 5. I also ran into this problem, trying to print something with 9 filaments on my A1 with a single AMS.
>         
>         
>         A1打印机已经支持了这个功能，请升级到最新的A1打印机固件，发送打印的时候在studio上勾选”外挂多色打印“的选项

That works for the external spool, if you use more than 1 colour, but not the AMS - and with the AMS and this, you don't have to do the manual switch between each colour change, only a couple

SaltWei, 2026-10-21:
> 7\. That works for the external spool, if you use more than 1 colour, but not the AMS - and with the AMS and this, you don't have to do the manual switch between each colour change, only a couple

搭配AMS使用也是可以的。
有AMS的时候，部分料选择AMS中的，部分料选择EXT就能开启AMS+外挂混打，外挂可以打印多个颜色的耗材，外挂耗材之间会自动暂停便于用户更换。ams内的耗材则自动更换

[image]

SaltWei, 2026-10-21:
需要注意的是，这个功能依赖打印机固件，且目前只有单头打印机支持这个功能。

DanaLeeTech, 2026-10-23:
> 8\. > 7. That works for the external spool, if you use more than 1 colour, but not the AMS - and with the AMS and this, you don't have to do the manual switch between each colour change, only a couple
>         
>         
>         搭配AMS使用也是可以的。
>         有AMS的时候，部分料选择AMS中的，部分料选择EXT就能开启AMS+外挂混打，外挂可以打印多个颜色的耗材，外挂耗材之间会自动暂停便于用户更换。ams内的耗材则自动更换
>         <img alt="Image" width="750" height="739" src="https://private-user-images.githubusercontent.com/110660407/679887538-7b42451e-4acc-402c-9bf7-3c61e93d4216.webp?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTI3Mzk1NTQsIm5iZiI6MTc5MjczOTI1NCwicGF0aCI6Ii8xMTA2NjA0MDcvNjc5ODg3NTM4LTdiNDI0NTFlLTRhY2MtNDAyYy05YmY3LTNjNjFlOTNkNDIxNi53ZWJwP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjEwMjNUMDcwNzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGYyN2U0NDc3MjkyOGM5MTRjOTUyNDdmZTk0YzAwOWRjOWM2NGVkODVlM2JmZDZmZDQ0ZGI3NGZjMWJiMWYwOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmcmVzcG9uc2UtY29udGVudC10eXBlPWltYWdlJTJGd2VicCJ9.q7baMfyqxExZKbR91pxCOU-I5XIh_JsZSEJvaGbU634">

I didn't want to use the external spool at all, and that would require a few more pauses than I had to for 9 colours.....I only had to pause it twice per layer, and not all of them would have been swapped both times.
I already knew that external spool swap pause was in there, but knew it doesn't work with the AMS alone, so I found another way 🙂