bambulab/BambuStudio #88: [Feature request] Velocity Printing aka "Tattooing"

hutje, 2022-08-01:
By varying the speed/extrusion volume at certain points it will create a subtle visible pattern or picture.
When overlaying a grayscale picture and placing it on the outer wall of the print, it will "wrap" itself around the object and will make a barely noticeable "indentation".

This would be ideal for your brand name, pattern or recycling icon. Also with printing lithophanes or vase mode this feature would be really nice.
[image]

Source: https://www.velocitypainting.xyz/blog/

wuyifanfancy, 2022-08-01:
It is a pretty cool feature. We will test and consider adding it to our plan list.

psiberfunk, 2022-08-06:
Wow that IS actually a pretty unique feature.  Would love to see that.