bambulab/BambuStudio #89: [BUG / 01.01.00.14 / macOS] Can't download network plug-in

jrahaim, 2022-08-02:
M1 Mac Pro. Monterey 12.4

Steps to reproduce. I
1) Install BambuStudtio
2) Launch
Try downloading the Plugin either from the initial setup, or from the banner top left. 
Download Failed.

There is no ~/Library/Application\ Support/BambuStudio/plugin directory just
log              
ota              
system    
user
       
[debug_Mon_Aug_01_21_58_40.log.0.txt](https://github.com/bambulab/BambuStudio/files/9238480/debug_Mon_Aug_01_21_58_40.log.0.txt)

jrahaim, 2022-08-02:
I double checked and tried a full re-install (removing ~/Library/Application\ Support/BambuStudio). 

Firewall off, no Antivirus installed.

Sirwill1968, 2022-08-02:
Do you by chance have a case sensitive file system on your Mac?

jrahaim, 2022-08-02:
> Do you by chance have a case sensitive file system on your Mac?

No

APFS.
- Not Encrypted. 
- Not Case Sensitive (just checked the partition info).

lanewei120, 2022-08-03:
from the log, it should be a network issue

I have just downloaded the plugin with the same address, it can success currently
could you also have a try again?

if still fails, we need to debug on the networking firstly

jrahaim, 2022-08-03:
@lanewei120 I have tried again (several times). Still fails. Works on a windows machine on the same network. 

Let me know what I can do to help debug networking.

lanewei120, 2022-08-04:
> @lanewei120 I have tried again (several times). Still fails. Works on a windows machine on the same network.
> 
> Let me know what I can do to help debug networking.

ok, could you help to upload the log both for windows and macos?
we will have a check firstly

thanks

lanewei120, 2022-08-04:
@jrahaim 
also could you give me an email?
i will send a debug dmg to you with more logs

thanks

jrahaim, 2022-08-04:
Ping works.
a now gone log (I tried a full reinstall this morning and forgot to save the log) included an attempt to download 
https://upgrade-file.bambulab.com/studio/plugins/01.01.00.11/mac_01.01.00.11.zip

wget of that file on the mac was successful. 

Here is a log from a fresh install (deleting ~/Library/Appliction_support/BambuLab on the mac
[MAC_debug_Thu_Aug_04_11_11_37.log.0.txt](https://github.com/bambulab/BambuStudio/files/9260919/MAC_debug_Thu_Aug_04_11_11_37.log.0.txt)

Now I can't seem to login from BambuStudio. Even after another fresh install (deleting ~/Library/Appliction_support/BambuLab)
Either clicking login doesn't work. or it hangs after I try to log in. 

Windows logs from initial installation
[Win_debug_network_Wed_Aug_03_15_47_05.log.0.txt](https://github.com/bambulab/BambuStudio/files/9260968/Win_debug_network_Wed_Aug_03_15_47_05.log.0.txt)
[win_debug_Wed_Aug_03_15_47_05.log.0.txt](https://github.com/bambulab/BambuStudio/files/9260969/win_debug_Wed_Aug_03_15_47_05.log.0.txt)

I will try to find your email address.

Sirwill1968, 2022-08-04:
You can create a ticket by emailing A brief overview to support@service.bambulab.com And that Lane Wei asked you to contact him directly.  That will allow him to work with you through support.   Reference this Github issue #89 in the ticket.

lanewei120, 2022-08-05:
> Ping works. a now gone log (I tried a full reinstall this morning and forgot to save the log) included an attempt to download [...]

I have sent the image to you in email
please check that

jrahaim, 2022-08-05:
installed. and tested.. Download still failed. Log sent.

Manually putting the zip contents in  ~/Library/Appliction_support/BambuLab/plugins worked with the version you sent.

alsandiego, 2022-08-09:
have the same issue.. using M1 mac and I have tried multiple vpns to see if its network error and same thing.. sometimes it will get 5% but it will still fail. I have upgraded to 1.1.1.1 version and still the same..
Can this plug-in not just be integrated on the App?

[debug_Tue_Aug_09_11_44_49.log.0.txt](https://github.com/bambulab/BambuStudio/files/9293864/debug_Tue_Aug_09_11_44_49.log.0.txt)

jrahaim, 2022-08-09:
@alsandiego I just sent them detailed logs for this on 1.1.1.1

To get up and running you can  wget the file 
wget https://upgrade-file.bambulab.com/studio/plugins/01.01.01.01/mac_01.01.01.01.zip

and unzip the contents into 
~/Library/Application Support/BambuStudio/plugins

let me know if you need details.

alsandiego, 2022-08-09:
I was able to get it going but after the login, the program just hangs as long as there is an internet connection..
I tried multiple connections and the same thing, it will hang when it starts(I think its trying to connect to something but then after a few seconds) it will work but if you ever select the Home or Printer tab, it will hang again.
I have ping api.bambulabs.com and its working fine.
[debug_Tue_Aug_09_12_42_00.log.0.txt](https://github.com/bambulab/BambuStudio/files/9294124/debug_Tue_Aug_09_12_42_00.log.0.txt)

lanewei120, 2022-08-10:
> have the same issue.. using M1 mac and I have tried multiple vpns to see if its network error and same thing.. sometimes it will get 5% but it will still fail. I have upgraded to 1.1.1.1 version and [...]

could you also help to try following commands and give the results:
ping api.bambulab.com

and more, there are some log named debug_network-xxx
please also help to upload that file here

thanks

alsandiego, 2022-08-11:
[Archive.zip](https://github.com/bambulab/BambuStudio/files/9309296/Archive.zip)
[result.txt](https://github.com/bambulab/BambuStudio/files/9309297/result.txt)
attached..

lanewei120, 2022-08-11:
> 
@alsandiego 
could you remove the plugins and try to download the plugin again, and then send the log to us?

also what is the current status of your Bambu Studio?
from the above log, 
it seems you haven't done something after the application launched

could you try to login, and go to the device tab, then send the log back to us?

thanks very much~

alsandiego, 2022-08-11:
I have removed and readd it.. when I load it, its blank screen and spinning ball for about 30sec then it loads but my printer is not visible and it shows I am logged in
[Archive 2.zip](https://github.com/bambulab/BambuStudio/files/9310202/Archive.2.zip)
n

lanewei120, 2022-08-12:
> I have removed and readd it.. when I load it, its blank screen and spinning ball for about 30sec then it loads but my printer is not visible and it shows I am logged in [Archive 2.zip](https://github.com/bambulab/BambuStudio/files/9310202/Archive.2.zip) n

Hello

we found some timeout and networking failure in your log
we have updated a network module at cloud side

Could you update the networking plugins firstly and then try again(launch studio, login if required, and goto the device tab)
if still failed, please help to send the log back to us

you can use the new plugins by 'wget https://upgrade-file.bambulab.com/studio/plugins/01.01.01.02/mac_01.01.01.02.zip'

sickerin, 2022-08-12:
Hi, I wasn't able to download the plugins. But I'm able to wget the plugins, and moved them to the folder. However, now the issue is that the login page is blank.
Device: Macbook Pro 16in 2019. macOS Monterey 12.4
[image]
[image]

lanewei120, 2022-08-12:
> Hi, I wasn't able to download the plugins. But I'm able to wget the plugins, and moved them to the folder. However, now the issue is that the login page is blank.

> Hi, I wasn't able to download the plugins. But I'm able to wget the plugins, and moved them to the folder. However, now the issue is that the login page is blank. Device: Macbook Pro 16in 2019. macOS Monterey 12.4 [image] [image]

Could you help to send the log in the folder
'~/Library/Application\ Support/BambuStudio/log' to us

thanks

sickerin, 2022-08-12:
[log.zip](https://github.com/bambulab/BambuStudio/files/9315073/log.zip)

lanewei120, 2022-08-12:
> 

could you give us an email?
we will send a debugging dmg to check the reason of why networking not work at your side

thanks

lanewei120, 2022-08-12:
> @lanewei120 Here [log.zip](https://github.com/bambulab/BambuStudio/files/9315073/log.zip)

@sickerin 
also, please help to try following things:
1. use your brower to access 'https://bambulab.com/en-GB/sign-in' to check whether it is accessible or not
2. ping 'api.bambulab.com' to check whether it is accessible or not
3. check the firewall setting whether BambuStudio is blocked or not

thanks

alsandiego, 2022-08-12:
still the same issue for me
[Archive.zip](https://github.com/bambulab/BambuStudio/files/9316905/Archive.zip)

TheJTizzle, 2022-08-12:
Seeing the same behavior as well. Happy to help submit any info to help out if needed.

lanewei120, 2022-08-12:
@sickerin @alsandiego @TheJTizzle 

if possible, please help to do following steps:
1. remove the plugins folder
2 execute the following command before launching the Bambu Studio (in terminal)
sudo tcpdump -s 0 -w test.pcap host [api.bambulab.com](http://api.bambulab.com/)
3. launch Bambu Studio, and install the plugin again
4. close Bambu Studio
5. send the following files back to us
test.pcap   //which is generated by tcpdump
three debug logs in the folder /Users/[user.name](http://user.name/)/Library/Application Support/BambuStudio/log
[image]

As it is a networking related issue only found on Mac in some Regions
Any ideas are welcome~

Regards
Lane

TheJTizzle, 2022-08-12:
Followed instructions but the only thing that was generated was this:

[debug_Fri_Aug_12_11_44_20.log.0.zip](https://github.com/bambulab/BambuStudio/files/9323460/debug_Fri_Aug_12_11_44_20.log.0.zip)
.

alsandiego, 2022-08-12:
> Followed instructions but the only thing that was generated was this:
> 
> [debug_Fri_Aug_12_11_44_20.log.0.zip](https://github.com/bambulab/BambuStudio/files/9323460/debug_Fri_Aug_12_11_44_20.log.0.zip) .

test.pcap is on your user profile

alsandiego, 2022-08-12:
[Archive 2.zip](https://github.com/bambulab/BambuStudio/files/9324368/Archive.2.zip)

I have tried different locations using VPN and same issue.

TheJTizzle, 2022-08-12:
[test.pcap.zip](https://github.com/bambulab/BambuStudio/files/9324677/test.pcap.zip)
test.pcap attached

lanewei120, 2022-08-15:
@sickerin @alsandiego @TheJTizzle @jrahaim 
we add a tool in bambu studio to test the networking issue
it can be download here
https://www.dropbox.com/s/1k5b8crn9q74y61/Bambu_Studio_mac-v01.01.01.01-20220815170132.dmg?dl=0

Could you help to try this version to use the tool for the networking test?
[image]
[image]

after that, please give the screen snapshot with the result to us
and then close the application, also send the log back to us

thanks very much

Lane

TheJTizzle, 2022-08-15:
Mon Aug 15 08:18:27:test bing start…
Mon Aug 15 08:18:27:test bambulab start…
Mon Aug 15 08:18:27:test storage start…
Mon Aug 15 08:18:27:test storage upgrade start…
Mon Aug 15 08:18:27:test storage download start…
Mon Aug 15 08:18:27:test http start…
Mon Aug 15 08:18:27:please install network module first
Mon Aug 15 08:18:27:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.01.00
Mon Aug 15 08:18:27:test storage download failed
Mon Aug 15 08:18:27:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Mon Aug 15 08:18:27:test storage download failed
Mon Aug 15 08:18:27:test bambulab failed
Mon Aug 15 08:18:27:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Mon Aug 15 08:18:27:test bing failed
Mon Aug 15 08:18:27:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Mon Aug 15 08:18:27:test storage failed
Mon Aug 15 08:18:27:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Mon Aug 15 08:18:27:test storage upgrade failed
Mon Aug 15 08:18:27:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]

JT

jrahaim, 2022-08-15:
[image]

Full text of log:
```
Mon Aug 15 08:21:01:test bing start…
Mon Aug 15 08:21:01:test bambulab start…
Mon Aug 15 08:21:01:test http start…
Mon Aug 15 08:21:01:test storage start…
Mon Aug 15 08:21:01:test storage upgrade start…
Mon Aug 15 08:21:01:test storage download start…
Mon Aug 15 08:21:01:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.01.00
Mon Aug 15 08:21:31:test bing failed
Mon Aug 15 08:21:31:test bambulab failed
Mon Aug 15 08:21:31:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 08:21:31:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 08:21:31:test storage upgrade failed
Mon Aug 15 08:21:31:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 08:21:31:test http failed
Mon Aug 15 08:21:31:test http failed, status = 0, error = 
Mon Aug 15 08:21:31:test storage download failed
Mon Aug 15 08:21:31:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 08:21:31:test storage download failed
Mon Aug 15 08:21:31:test storage failed
Mon Aug 15 08:21:31:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10004 milliseconds
[Error 28]
```

If I wget the file from the command line
`% wget https://api.bambulab.com/v1/iot-service/api/slicer/resource\?slicer/plugins/cloud\=01.01.01.00`

I get this:
```
--2022-08-15 08:23:16--  https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.01.00
Resolving api.bambulab.com (api.bambulab.com)... 47.88.61.230, 47.254.16.94
Connecting to api.bambulab.com (api.bambulab.com)|47.88.61.230|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1764 (1.7K) [application/json]
Saving to: ‘resource?slicer%2Fplugins%2Fcloud=01.01.01.00’

resource?slicer%2Fplugins%2Fcloud=01.01.01.00                   100%[====================================================================================================================================================>]   1.72K  --.-KB/s    in 0s

2022-08-15 08:23:30 (280 MB/s) - ‘resource?slicer%2Fplugins%2Fcloud=01.01.01.00’ saved [1764/1764]
```

jrahaim, 2022-08-15:
Note: Either the wind is blowing just right, OR they fixed a network issue or these steps I have taken are making everything run much smoother (it's useful). Using the version provided above (I didn't try any others yet). 

1) from ~/Application Support/BambuStudio
`wget https://upgrade-file.bambulab.com/studio/plugins/01.01.01.02/mac_01.01.01.02.zip`
`mkdir plugins;cd plugins`
`unzip ../mac_01.01.01.02.zip`
2) adding 47.254.16.94	api.bambulab.com to /etc/hosts  (you will need to sudo, or use some other root command to add it). 
example 
`sudo vi /etc/hosts`
(but only if you know VI).
3) Start  bambu studio I can see my device and watch video and the app isn't stalling / hanging like it was.

alsandiego, 2022-08-15:
Mon Aug 15 07:23:47:test bing start…
Mon Aug 15 07:23:47:test bambulab start…
Mon Aug 15 07:23:47:test http start…
Mon Aug 15 07:23:47:test storage start…
Mon Aug 15 07:23:47:test storage download start…
Mon Aug 15 07:23:47:test storage upgrade start…
Mon Aug 15 07:23:47:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.01.00
Mon Aug 15 07:23:47:test bambulab ip resolved = 47.254.16.94
Mon Aug 15 07:23:47:test bambulab ok
Mon Aug 15 07:23:47:test http ok
Mon Aug 15 07:24:17:test bing failed
Mon Aug 15 07:24:17:test storage failed
Mon Aug 15 07:24:17:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 07:24:17:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 07:24:17:test storage upgrade failed
Mon Aug 15 07:24:17:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10003 milliseconds
[Error 28]
Mon Aug 15 07:24:17:test storage download failed
Mon Aug 15 07:24:17:status=0, body=, error=curl:Timeout was reached:
Resolving timed out after 10001 milliseconds
[Error 28]
Mon Aug 15 07:24:17:test storage download failed

lanewei120, 2022-08-16:
@sickerin @alsandiego @TheJTizzle @jrahaim
thanks for the result

if execute "ping api.bambulab.com" before launch the Bambu Studio,
is the network functions working in Bambu Studio?

alsandiego, 2022-08-16:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Yes

lanewei120, 2022-08-16:
> Mon Aug 15 08:18:27:test bing start… Mon Aug 15 08:18:27:test bambulab start… Mon Aug 15 08:18:27:test storage start… Mon Aug 15 08:18:27:test storage upgrade start… Mon Aug 15 08:18:27:test storage [...]

@TheJTizzle 
it seems your error reason is different:
"Couldn’t connect to server:
[Error 7]"

it maybe caused by the firewall setting,
could you check the firewall setting on your mac firstly?
[image]

TheJTizzle, 2022-08-16:
> … <x-msg://8/#>

It’s not the firewall, it appears to be a block caused by Adguard. https://adguard.com <https://adguard.com/>. If I disable the app it allows the download. I tried adding the app and the API link to the allow lists but it still gets block, so still wanna look into it, but I found the source for you guys and am functional while I wait for my printer!

JT

alsandiego, 2022-08-19:
figured out the issue on my end.. I have Cisco Anyconnect VPN installed and even though it is not connected, it has a network device that it installs that continuously runs on any connected network.. after uninstalling it, everything started working now

jrahaim, 2022-08-20:
That seemed to have fixed my issues as well!

docangle, 2022-08-20:
I m having the same problem, but I don't have adgaurd installed, her is my log:

Sat Aug 20 08:32:52:test bing start…
Sat Aug 20 08:32:52:test bambulab start…
Sat Aug 20 08:32:52:test http start…
Sat Aug 20 08:32:52:test storage start…
Sat Aug 20 08:32:52:test storage upgrade start…
Sat Aug 20 08:32:52:test storage download start…
Sat Aug 20 08:32:52:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.01.01.00
Sat Aug 20 08:32:52:test bambulab ip resolved = 47.88.61.230
Sat Aug 20 08:32:52:test bambulab ok
Sat Aug 20 08:32:52:test http ok
Sat Aug 20 08:32:53:test bing ip resolved = 204.79.197.200
Sat Aug 20 08:32:53:test bing ok
Sat Aug 20 08:32:55:test storage failed
Sat Aug 20 08:32:55:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Sat Aug 20 08:32:55:test storage upgrade failed
Sat Aug 20 08:32:55:status=0, body=, error=curl:Couldn’t connect to server:

[Error 7]
Sat Aug 20 08:32:56:downloading 10%
Sat Aug 20 08:32:56:downloading 20%
Sat Aug 20 08:32:56:downloading 30%
Sat Aug 20 08:32:56:downloading 40%
Sat Aug 20 08:32:56:downloading 50%
Sat Aug 20 08:32:56:downloading 60%
Sat Aug 20 08:32:56:downloading 70%
Sat Aug 20 08:32:56:downloading 80%
Sat Aug 20 08:32:56:downloading 90%
Sat Aug 20 08:32:57:downloading 100%
Sat Aug 20 08:32:57:test storage download ok
Sat Aug 20 08:32:57:test storage download ok

lanewei120, 2022-08-22:
> I m having the same problem, but I don't have adgaurd installed, her is my log:
>
> Sat Aug 20 08:32:52:test bing start… Sat Aug 20 08:32:52:test bambulab start… Sat Aug 20 08:32:52:test http start… [...]

@docangle 
Your error code is "curl:Couldn’t connect to server: [Error 7]"
it is very likely caused by your networking environment

could you try following things firstly
1. check whether the firewall is turned on or not
if yes, make sure BambuStudio is not blocked by firewall
2. check whether there are some software installed which will block Bambu Studio
such as "Cisco AnyConnect" mentioned above
3. try to ping following address and check whether they are successful or not
ping api.bambulab.com
ping upgrade-file.bambulab.com
ping upload-file.bambulab.com

EmbraceNext, 2022-08-22:
For anyconnect users,  I have found this is a work around instead of uninstalling.  Since anyconnect installs socket filters, just go into your network properties and make them inactive until you need to connect to your VPN.  
[image]

lanewei120, 2022-08-23:
It seems "Cisco Anyconnect Socket Filter" blocks the DNS probe in these cases
just found some solutions from internet(either of them is ok)
1. use following commands in the terminal to disable the "Cisco Anyconnect Socket Filter"
/Applications/Cisco/Cisco\ AnyConnect\ Socket\ Filter.app/Contents/MacOS/Cisco\ AnyConnect\ Socket\ Filter -deactivateExt

2. disable/inactive the "Cisco Anyconnect Socket Filter" in the Network page from "System Preferences"
just the same as what @EmbraceNext mentioned above

3. uninstall  "Cisco Anyconnect Socket Filter"

4. when install Cisco AnyConnect, just installed the component "VPN" (not confirmed whether it is ok or not)
 [image]

docangle, 2022-08-23:
I don’t have Anyconnect installed.  The issue is iCloud Private Relay which is part of iCloud +. I have to keep it disabled since it is masking IP Address._______________Marshall Burke

SilverApe, 2022-09-06:
What helped me and @EmbraceNext also mentioned it is to make the network connections inactive.
[image]

lanewei120, 2022-10-14:
as these issues have been solved, close it

closed 2022-10-14 by lanewei120