bambulab/BambuStudio #90: Generic PETG Not printable via AMS (mac client)

sleeper-seth, 2022-08-02:
After setting up a generic PETG into the AMS, it is not possible to select that filament within the Print window on the Mac client

Wild Guess: It appears like the client is looking for "PET" and the AMS supplies "PET.G" as the type.

Repro Steps:

1. Setup Generic PETG on X1C in AMS slot 4
2. Setup new print. Select AMS material 4 (Generic PETG) OR Select Generic PETG (both fail the same)
3. Slice
4. Print button to bring up Print window
5. Label "PET.G" is not selectable in AMS slot 4

Attempting to print anyways will start the print with AMS slot 1 using PETG settings (regardless of PLA being in slot 1)

wuyifanfancy, 2022-08-04:
The issue has been fixed and will be in the next release version.

YifanWuBambu, 2022-08-09:
Fixed in the latest version.

closed 2022-08-09 by YifanWuBambu