bambulab/BambuStudio #91: No warning given when Flow Calibrations intersect with models

LegacyOfBob, 2022-08-03:
After placing a large model, or a model that otherwise overlaps with the flow calibration areas on the build plate, and performing slicing, no warning is given that an overlap has occurred. This leads to models being printed where the flow calibrations are welded in and need to be cut away. The slicer should at least present a warning to the user before printing. 

(I am aware the auto-arrange feature has an option to avoid the area.)

wuyifanfancy, 2022-08-04:
We will add this warning in recent release.

tsdexter, 2022-08-15:
@wuyifanfancy is it possible to move the flow calibration somewhere else so we can still print models such as a large diameter circle and also calibrate by having the pattern in the hollow center of the circle?