bambulab/BambuStudio #92: [Feature Request] Surface Mode (not the same thing as Spiral Mode)

Panacaguy, 2022-08-03:
Please consider adding a feature like the "Surface Mode" in Cura (within the Special Modes section) or the "Single Shell Surface Mode" in Ideamaker (within the Layer tab). This is not the same thing as Spiral Mode, which is already implemented. Many 3D printable airplane designs require this Surface Mode feature to properly slice with only a single outside wall and no top or bottom surface. It allows non-manifold geometry to be printed and differs from spiral mode in that it can print both inside and outside surfaces that are not one continuous line. Using the Spiral Mode feature in Bambu Studio does not give the correct results.

I submitted this once about a week ago (see the Closed section) but it was closed with the response that it is already implemented as Spiral Mode. See the comments on that thread. Spiral Mode and Surface Mode are not the same thing. Please look into the differences and implement into Bambu Studio.

randyre, 2022-10-21:
The sooner we can get this the better, Would like to start printing some of my airplane parts. really don't want to have to resurect my old CR-10 just for the planes.

SaltWei, 2022-12-16:
Even/Odd setting has been added in latest 1.4 version

closed 2022-12-16 by SaltWei