bambulab/BambuStudio #98: Not running on either of two windows computers

melfahn, 2022-08-05:
I have had no luck installing/running this on two Windows machines, one Windows 11, one Windows 10.  Windows 11 installed fine but won't allow me to login "internal server error". Windows 10 installs (without the network plug-in that the Windows 11 prompted for). However it will not start. Shows splash and then A white UI for a brief moment and then it crashes. Thank you. The log initially showed that it couldn't find the network DLL. I copied out over from the Windows 11 machine, but it didn't make a difference except for the log getting a little further. Thank you

[Windows 10 crashing.zip](https://github.com/bambulab/BambuStudio/files/9273935/Windows.10.crashing.zip)
[Windows 11 not allowing login-internal server error.zip](https://github.com/bambulab/BambuStudio/files/9273944/Windows.11.not.allowing.login-internal.server.error.zip)

melfahn, 2022-08-06:
the Windows 11 not logging in issue appears to have been temporary. This morning it allowed me to login.

FiorenzaPrime, 2022-08-07:
I'm having the same issue with the program failing to start on Windows 10. The splash screen appears followed by a blank window, then the application force closes.

[image]
[image]
[eventviewer log.txt](https://github.com/bambulab/BambuStudio/files/9277057/eventviewer.log.txt)
[system info DxDiag.txt](https://github.com/bambulab/BambuStudio/files/9277058/system.info.DxDiag.txt)

FiorenzaPrime, 2022-08-15:
Updated to the newest version 01.01.01.01 and it's working great now.

lanewei120, 2022-08-23:
@melfahn 
do you still have issue now?

we have fixed a crash issue after splash window

closed 2022-09-21 by lanewei120