bambulab/BambuStudio #101: Screen stuck at "Connecting to printer" but i can hit "send"

pitops, 2022-08-07:
Hey so i think this counts as a bug. In the picture below you can see that it says `Connecting to printer` but in reality i can send the model over and starts printing.

[image]

However, I get no info from the printer on the device screen but if i hit the play button to see the camera, again it works. This stopped working like 2 weeks ago second day i received the printer.

Any clues?

pitops, 2022-08-12:
Hey all after updating to v1.1.1.1 i don't have this issue, yet. Do what you will with this information ;p

YifanWuBambu, 2022-08-31:
Hi @pitops 
We optimized the network connection in v1.1.1.1.
Have you still met this issue recently?

pitops, 2022-08-31:
Hi @YifanWuBambu i think its fixed, no issues so far

closed 2022-08-31 by pitops