bambulab/BambuStudio #102: Auto arrange ignores available area

pitops, 2022-08-07:
Below is how i actually placed the tiles to print them

[image]

However, if i hit 'auto arrange' it does this (played with almost all variations of settings)

[image]

Result of above settings

[image]

The setting that did work are only the following

[image]

[image]

In my opinion it should be able to arrive in this kind of setup with any setting in there. There is plenty of space to arrange the objects

tangketan, 2022-08-10:
Can't reproduce the bug on version 1.1.1.1 (the latest). Would you please try this version, and kindly provide the 3mf file if the bug is still there?

Allram, 2022-08-12:
It's the same on 1.1.1.1, attached the .3mf file. It covers the "unprintable" area when using autoarrange.
[image]

.3mf file: https://www.transfernow.net/dl/20220812bcaSxHBO

tangketan, 2022-08-31:
The file can't be downloaded. This issue should have been solved, I can't reproduce it with similar models on 1.1.1.4.

closed 2022-10-14 by YifanWuBambu