bambulab/BambuStudio #103: [BUG / 01.01.00.14 / Windows] Support filament disappear when Top Z distance is set to 0mm

thrutheframe, 2022-08-08:
Layer height is set to 0.2mm and Top Z distance is set to 0mm for support filament, the support layer disappears. See images. 

It happens only to 0.2mm layer height. 0.12, 0.16, 0.24 all ok

With 0.2mm Top Z distance:
[image]

With 0.0mm Top Z distance:
[image]

thrutheframe, 2022-08-09:
@YifanWuBambu I also checked with the newer ver 1.1.1.1. Issue still persist. thanks

YifanWuBambu, 2022-08-09:
@thrutheframe 
The color scheme is set to Filament if more thant one filament is used in selected plate.
Please change the color scheme to Line Type to see the support feature more clearly.
[image]

closed 2022-08-31 by YifanWuBambu