bambulab/BambuStudio #105: Feature Request: QoL requests

kuljaden, 2022-08-08:
First off, thank you Bambu team for everything that you've done so far! This is just a small QoL request.

In the Bambu Labs slicer, the first request is to add an option to Center to Build plate. 
Second QoL feature request is to add a bounding box to the slicer, this is mostly for the Z height so we can have a visual representation of the height vs relying on the Z value.
Third QoL feature request is to have an Auto Slicer toggle. A small QoL request to allow the model to auto slicer if there are any changes and allows us to now rely on having to manually click it or toggling between views.
Fourth Request is to have a play feature. What I mean by this is Cura has an option where you can click play and the slicer will play a virtual representation of how the extruder will move on the slicer on that layer, and when the layer is done it continues to play the next layer. This could be useful to see how the extruder will behave during specific scenarios.

Thank you!

YifanWuBambu, 2022-08-09:
Hi @kuljaden , thanks for your suggestions. We will add them to the plan list.

suit, 2023-03-02:
this can maybe pulled from a global database that is could synced with Bambu Cloud

for example the german manufacturer Das Filament provides profiles for at least 2 of their filaments which unfortunately need to be imported manually into the AppData directory

https://www.dasfilament.de/blog/filamentprofile-fuer-den-bambu-lab-p1p-bambu-studio

there are lots of filament manufacturers that test their filaments too - it would be beneficial, if those can be included directly into a cloud solution provided by bambu labs so you can search online directly bambu studio for either user created profiles or manufacturer created profiles

this also can lead to a crowd sourcing solution where you can enter data that work for you (brand, material, color etc.) and bambu studio creates an "average" recommendation for those filaments