bambulab/BambuStudio #106: Feature Request: "Avoid crossing perimeters"

cdsmakestuff, 2022-08-08:
Please add the "Avoid crossing perimeters" feature / switch to the available settings.

YifanWuBambu, 2022-08-09:
Hi @cdsmakestuff ,
Did there any printing problem happen with current settings?

stephenjamieson, 2022-08-21:
This should help with PETG stringing

psiberfunk, 2022-08-22:
This is a very useful setting in PrusaSlicer/Superslicer which tends to significantly reduce wall artifacts .  I almost always leave it on by default .. but it would be good to know if Bambu has this on or not ?

schreiaj, 2022-09-14:
@YifanWuBambu yes. The artifacts are most prevalent with tpu filaments as they tend to be stringy under the best of circumstances. 

[image]

The wisps are all travel moves across boundaries. By disallowing them they are sealed inside the part. 

While printing cooler may reduce some artifacts (specifically with petg) it also reduces layer adhesion reducing the strength of the part.

psiberfunk, 2022-09-14:
TPU is absolutely a case where I once spent days tuning for optimal settings and having this setting on was one of the most important .  I have an army of Evee prints to prove it too !

[image]

psiberfunk, 2022-10-04:
Any likelyhood we will get this one implemented soon, seeing as it's already in the Prusa code?

YifanWuBambu, 2022-10-09:
Thanks for the information.
We will add this feature in our recent release.

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu

Bpositivedotdk, 2022-10-31:
Good to see some combing. But, it is placing the travel moves at the edge. Would it be possible to move these travels into the model as in Cura. With 2-walled perimeters it should be placed in the middle!

tokamac, 2022-10-31:
> Good to see some combing. But, it is placing the travel moves at the edge. Would it be possible to move these travels into the model as in Cura. With 2-walled perimeters it should be placed in the middle!

I don't think it's possible unless Bambu Lab implements a whole new code for that. Yet, BambuStudio is not a brand new thing: it's a fork of PrusaSlicer. And PrusaSlicer does not have the fine optional tuning of Cura's combing, sadly.

Bpositivedotdk, 2022-10-31:
I never worked with prusaslicer.... but I'm definitely missing all the many settings of Cura. - Will there soon be a working profile for Cura!!!? :-)

knbpixels, 2023-01-03:
So the "avoid crossing walls" feature was added but does not actually work or change the path of the hot end.

psiberfunk, 2023-01-03:
> So the "avoid crossing walls" feature was added but does not actually work or change the path of the hot end.

Do you have an example to show this ?

knbpixels, 2023-01-03:
> > So the "avoid crossing walls" feature was added but does not actually work or change the path of the hot end.
> 
> Do you have an example to show this ?

Yes, in the file I've been trying to print.  I get constant travel moves across walls regardless of if the setting is on or not.  I will attach the file.  As an example, examine layer one, the part of the print with the "98" lettering.
[radio face number block print file.3mf.zip](https://github.com/bambulab/BambuStudio/files/10339066/radio.face.number.block.print.file.3mf.zip)

tokamac, 2023-01-03:
You should check in the slicer before printing. I see clear differences in travel moves in the preview pane of BambuStudio when Avoid crossing wall is enabled or disabled. Here are some examples with the simple STL of a bowl: 

**Seam position : aligned; Avoid crossing wall disabled:**
[image]

**The same when enabling Avoid crossing wall:**
[image]

and:

**Seam position : random; Avoid crossing wall disabled:**
[image]

**The same when enabling Avoid crossing wall:**
[image]

I think you are mistaking two different features and I agree that Bambu Lab's wording for the feature is misleading. But it's not entirely their fault as they are taking PrusaSlicer's "Avoid Crossing perimeters" which does the exact same thing: 
https://help.prusa3d.com/article/layers-and-perimeters_1748#avoid-crossing-perimeters

BambuStudio's "Avoid crossing wall" feature and PrusaSlicer's "Avoid crossing perimeters" are if fact much like Cura's "Combing" feature (within Infill). _Combing_ in Cura keeps the nozzle within already printed areas when traveling (it does not prevent the nozzle from crossing walls… just that maintaining the nozzle over already printed walls will necessarily reduce the number of switches from one wall to another).

In fact it seems that people are expecting a different feature that also exist in Cura, which is called "Avoid Printed Parts When Traveling". But this is not the feature we are talking about here… even if its name suggests the contrary.

knbpixels, 2023-01-03:
I'm familiar with "combing" in Cura, and that's exactly the behavior I would expect to see in my file.   Not sure why you think I'm not checking the slicer before I print...I'm definately doing that.  To be clear, what I'm expecting the setting to accomplish is that once perimeter walls are laid down travel only moves will occur within walls or infill, and not cross into open space unless necessary.   And yet this is the behavior I'm seeing:
[image]

psiberfunk, 2023-01-03:
If you look VERY carefully at that "crossing", you'll actually see it's a travel AFTER a z-hop, which is not really considered a crossing, since it's at a different Z-height when it's making the move.  Take a look at this little video to see what I mean. 

[Video.zip](https://github.com/bambulab/BambuStudio/files/10339547/Video.zip)

knbpixels, 2023-01-03:
> If you look VERY carefully at that "crossing", you'll actually see it's a travel AFTER a z-hop, which is not really considered a crossing, since it's at a different Z-height when it's making the [...]

That seems like a poor implementation.   It's not like a zhop prevents oozing, as is evident in my print where I have strings in these travel paths.  Shouldn't the travel still take place over the printed piece to reduce strings?

psiberfunk, 2023-01-03:
> > If you look VERY carefully at that "crossing", you'll actually see it's a travel AFTER a z-hop, which is not really considered a crossing, since it's at a different Z-height when it's making the move. Take a look at this little video to see what I mean.
> > [Video.zip](https://github.com/bambulab/BambuStudio/files/10339547/Video.zip)
> 
> That seems like a poor implementation. It's not like a zhop prevents oozing, as is evident in my print where I have strings in these travel paths. Shouldn't the travel still take place over the printed piece to reduce strings?
> 
> I still don't understand why there is no change in my slicer preview with the setting either on or off, and my gut tells me it's not a coincidence that its not doing anything

Actually, their z-hop implementation with a spiral lift is probably one of the better ones out there.. and i'm by no means a shill for Bambu.. they've got plenty of issues.  It may be the case that this particular print doesn't benefit a ton from this feature.  Here's an example where it really matters:  

With wall crossing "allowed":
[image]

With avoid crossing walls:  
[image]

psiberfunk, 2023-01-03:
Notice how the print time changes and the travels change a LOT

knbpixels, 2023-01-03:
> Notice how the print time changes and the travels change a LOT

Sure, and I don't mind the spiral hop at all, I just wish the the hot end would then travel over the print in my file.

psiberfunk, 2023-01-03:
> > Notice how the print time changes and the travels change a LOT
> 
> Sure, and I don't mind the spiral hop at all, I just wish the the hot end would then travel over the print in my file.

Yea, in the Prusa code and elsewhere i've seen this implemented it's universally understood to not be "crossing" the perimeter when z-hopped.  I can't say I disagree with it.  In theory there could be an extra "also stay in perimeters when z-hopped".. but even the main free slicer with the most customization (either Cura or SuperSlicer, depending on who you ask), does not have this.